basler/pylon-ros-camera #2: Remove dependency on dragandbot_common

NikolasE, 2019-10-07:
Hey!

Including a company-specific package just for two tiny definitions looks like an overkill. Wouldn't it be easier to add them to the camera_control_msgs?

pablo-quilez, 2019-10-07:
Hi! drag&bot is an industrial framework and therefore needs to guarantee that all drivers are always working and not in error state. Therefore, the driver was adapted to publish its own status through our standardized status message, which is gather through a component manager. We cannot add it to the camera_control_msgs because many drivers use this definitions. As for us is a requirement to have the message, and because of now we are busy within a first-release phase, this will be kept at the moment as it is. Later we may remove it and keep it only in our fork https://github.com/dragandbot/pylon-ros-camera 
Thank you!

NikolasE, 2019-10-07:
Is the component manager something from the DNB-framework?

closed 2020-01-16 by pablo-quilez