basler/pylon-ros-camera #20: Any support for YUV colour format planned?

Crusty82, 2020-02-03:
Hi :)

When we moved from USB to GigE we lost the ability to use PGI on-board demoasicing because the GigE Ace U cams only support YUV as a debayered format. 
The stack here however does not support the YUV format.

Are there any plans to add those?

Thanks a lot :)

pablo-quilez, 2020-02-20:
Hello Crusty, 

ROS latest version Melodic seems not to support YUV format natively, at least according to the standard encodings found here http://docs.ros.org/melodic/api/sensor_msgs/html/image__encodings_8h_source.html 

I think it may not be possible to add this in the camera, at least without adding it first to the ROS vision stack. Please someone correct me if I am wrong or missing future stack improvements.

Cheers,
Pablo

pablo-quilez, 2020-04-16:
Hello @Crusty82 

I think we may add in the driver this format (8 bit):
const std::string YUV422="yuv422";

Would be this a solution to you? I can prepare a Pull Request for you to test it.

Crusty82, 2020-04-16:
Yep that would be great if I understand that correctly!

pablo-quilez, 2020-04-17:
Hello @Crusty82 

I am taking a look now to the issue. Which camera model are you using? Do you know exactly which format is the relevant in your case? Do your camera support YUV422_YUYV_Packed?

**1) Basler YUV suported formats (camera-depend) are**: YCbCr422_8, YUV422Packed, YUV422_YUYV_Packed, YCbCr420_8

YUV* formats are supported by the old cameras, but the newest models support only YCbCr422_8.
It seems that the format YUV422_YUYV_Packed is the one which ROS required. For YCbCr422_8 probably we will need some conversion algorithm, for example http://e2e.ti.com/support/processors/f/791/t/199637?YUYV-to-UYVY-formart-conversion-on-DM8168

**2) ROS seems to support natively the following packaged UYVY format:**

```
00098     // Miscellaneous
00099     // This is the UYVY version of YUV422 codec http://www.fourcc.org/yuv.php#UYVY
00100     // with an 8-bit depth
00101     const std::string YUV422="yuv422";
```

http://docs.ros.org/jade/api/sensor_msgs/html/image__encodings_8h_source.html

> YUV 4:2:2 (Y sample at every pixel, U and V sampled at every second pixel horizontally on each line). A macropixel contains 2 pixels in 1 u_int32.
> 
> UYVY format has a horizontal subsampling period of 2 for both the U and V components indicating that U and V samples are taken for every second pixel across a line. Their vertical subsampling period is 1 indicating that U and V samples are taken on each line of the image.

Crusty82, 2020-04-17:
Hi,

We are  using aca1920-40gc. It's not yet the Ace2 series, but they are not
old cameras.
In the end the goal is to have the cameras as specified above de-mosaic via
PGI and have the ROS stack support this.

According to the Basler docs those cams support:

acA1920-40gc
Mono 8
Bayer RG 8a
Bayer RG 12a
Bayer RG 12 Packeda
YUV 4:2:2 Packed
YUV 4:2:2 (YUYV) Packed

pablo-quilez, 2020-07-22:
Hi,

https://github.com/basler/pylon-ros-camera/tree/yuv422 we have created a branch to test the feature. It is possible that is still not working as expected because we tested it first with a modern camera which has YCbCr422_8 instead of YUV422_YUYV_Packed encoding. Until next week will be difficult for us to test it with a YUV422_YUYV_Packed compatible model. Sorry for the late reply, I hope the topic is still relevant to you and thanks for your understanding.

Regards,
Pablo

pablo-quilez, 2020-08-12:
Hi, 

after testing with different real hardware, it seems that is not working directly without a transformation. I tested the three formats YCbCr422_8, YUV422_YUYV_Packed and YUV422Packed and non of them were correctly displayed in rviz.

Basler format:
https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/how-does-the-yuv-color-coding-work/15182/

ROS: This is the UYVY version of YUV422 codec http://www.fourcc.org/yuv.php#UYVY with an 8-bit depth
http://www.fourcc.org/yuv.php#UYVY

It seems that a transformation from YCbCr422_8 or YUV422Packed to RGB / ROS yuv422 would be needed. According to the documentation, is an 8-bit UYVY implementation which in theory should be compatible with YUV422_YUYV_Packed but in the real tests it seems to be different. I think this issue would require much more effort, also considering that a transformation will affect the performance.

Regards

pablo-quilez, 2020-08-31:
Hi, 

The driver is now supporting YUV format as I merged into **devel** after testing it. In the next days I am planning to merge devel into master. 

The camera format YUV422Packed is the same UYVY format as the supported by ROS. U and V components are the color components and first and second Y correspond with the brightness of the first and second pixels. I had the chance to test the branch with the acA1440-73gc camera, so that I was able to fix some issues and finish the implementation.

image_view can display yuv422 format. RVIZ plugin doesn't support the format.
[image]

[image]

Cheers,
Pablo

closed 2020-08-31 by pablo-quilez

m-binev, 2020-08-31:
@Crusty82 Hi, check this out, please. I could fix it eventually! Cheers, Momchil