basler/pylon-ros-camera #24: Node is not writable when call start_grabbing

MomoHsieh, 2020-02-24:
camera: acA2440-75uc
OS/device: Ubuntu 16.04 with ROS(kinetic) in docker / Jetson AGX Xavier
Opened two terminal,
First called `roslaunch pylon_camera pylon_camera_node.launch`

```
PARAMETERS
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: 
 * /pylon_camera_node/enable_current_params_publisher: True
 * /pylon_camera_node/enable_status_publisher: True
 * /pylon_camera_node/frame_rate: 5.0
 * /pylon_camera_node/gige/mtu_size: 1500
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    pylon_camera_node (pylon_camera/pylon_camera_node)

auto-starting new master
process[master]: started with pid [1183]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to afe8bf76-56ad-11ea-82b5-00044bcc10e6
process[rosout-1]: started with pid [1196]
started core service [/rosout]
process[pylon_camera_node-2]: started with pid [1208]
[ WARN] [1582511462.441443735]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1582511462.441945201]: No Device User ID set -> Will open the camera device found first
[ INFO] [1582511462.731535945]: Found camera with DeviceUserID N/A: acA2440-75uc
[ WARN] [1582511462.986045287]: No User Set Is selected, Camera current setting will be used
[ INFO] [1582511462.992758423]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] 
[ WARN] [1582511462.993106632]: No image encoding provided. Will use 'mono8' or 'rgb8' as fallback!
[ WARN] [1582511463.128782253]: [] name not valid for camera_info_manager
[ INFO] [1582511463.220310896]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
[ WARN] [1582511463.220583774]: Will only provide distorted /image_raw images!
[ INFO] [1582511463.226793332]: Startup settings: encoding = 'mono8', binning = [1, 1], exposure = 3000, gain = 0, gamma = 1, shutter mode = default_shutter_mode
[ INFO] [1582511463.228155000]: Start image grabbing if node connects to topic with a frame_rate of: 5 Hz
[ INFO] [1582511463.229123785]: Camera not calibrated
[ INFO] [1582511463.277962964]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] 
[ WARN] [1582511463.386160986]: [] name not valid for camera_info_manager
[ INFO] [1582511463.453152433]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
[ WARN] [1582511463.453369308]: Will only provide distorted /image_raw images!
[ INFO] [1582511463.461328139]: Startup settings: encoding = 'mono8', binning = [1, 1], exposure = 3000, gain = 0, gamma = 1, shutter mode = default_shutter_mode
[ INFO] [1582511749.218811729]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] 
```

Second call `rosservice call /pylon_camera_node/start_grabbing`
```
root@AGX:~/catkin_ws# rosservice call /pylon_camera_node/start_grabbing
success: False
message: "Error"
```

and I got msg in first terminal
```
[ INFO] [1582511749.218811729]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] 
[ERROR] [1582511749.223653481]: An exception while setting target image encoding to 'mono8' occurred: Node is not writable
[ERROR] [1582511749.225676099]: Error while start grabbing
```

Can you help me to solve this issue?

pablo-quilez, 2020-04-09:
Hello,

it seems that it could be related to PixelFormat in the camera configuration. Could you please check with Pylon Viewer Basler application which configuration this parameter has?

The following code snippet seems to produce an exception:
```
if ( GenApi::IsAvailable(cam_->PixelFormat) )
{
  GenApi::INodeMap& node_map = cam_->GetNodeMap();
  GenApi::CEnumerationPtr(node_map.GetNode("PixelFormat"))->FromString(gen_api_encoding.c_str());
}
```
Also, can you check if the camera starts already automatically grabbing?

Thanks!

m-binev, 2020-04-21:
Hi @MomoHsieh ,

generally speaking, when StartGrabbing() is called once, all critical parameters that may change the buffer payload size (Pixel Format, Width, Height, etc.) become locked, i.e. RO.
Thus, when trying to set e.g. Pixel Format after StartGrabbing(), this will always result in an exception.
Thus, setting Pixel Format can only be done after calling StopGrabbing(). 
Regards

closed 2020-10-07 by pablo-quilez