basler/pylon-ros-camera #25: PTP synchronization with multiple cameras

AleRiccardi, 2020-02-26:
Hello,

the Basler [documentation](https://docs.baslerweb.com/precision-time-protocol.html) provide detailed information on how to synchronize multiple cameras using the PTP protocol, but the problem is that I cannot find the way to match does information for the Pylon ROS Driver.

I already have a network with a master clock and I enabled, through the Pylon Software Suite, the PTP clock synchronization for both cameras (as mentioned [here](https://docs.baslerweb.com/precision-time-protocol.html#enabling-ptp-clock-synchronization)).

What I am doing right now is running two different Pylon nodes specific to a singular camera and looking at the header of their topic in order to understand if the time stamp of the two cameras coincides.

Because this is not a professional way to check their synchronization and it doesn't give me the prove that they are really synchronized I would like:

* to know if there is a way to understand if the cameras are really synchronized with the master clock
* or to know how to synchronize them using the Pylon ROS driver.

All of this, if possible, only by using the Pylon ROS driver.

Thank you,
Alessandro

ctyfang, 2020-04-05:
From the user manual, it appears that you can check the synchronization offset from the master clock with some API commands.

"1.   Make sure that PTP clock synchronization has been enabled on all devices (see Section 8.8.1on page 268).

2.   Execute the GevIEEE1588DataSetLatch command to take a snapshot of the camera’s current PTP clock synchronization status.
3.   Read one or more of the following parameter values from the device:
GevIEEE1588OffsetFromMaster
GevIEEE1588ClockId
GevIEEE1588ParentClockId
GevIEEE1588StatusLatched

All of these parameter values relate to exactly the same point in time, i.e. the point in time when the device received the GevIEEE1588Latch command"

pablo-quilez, 2020-04-09:
Hello,

I am sorry that currently we may not address this topic for the Pylon ROS driver in a short time due to our resources limit. Anyway we keep it open. If someone wants to contribute in this topic it will be very welcomed!

Thanks!

FrancoisPicardDTI, 2022-07-12:
Hello

The ROS2 pylon driver v1.1.0 gives now access through services to PTP related parameters and commands, allowing to enable PTP synchronization, to issue scheduled or not action commands, to use the periodic signal camera feature for ACE2 and the synchronous free-run feature for ACE1.
I am thus closing this issue.

closed 2022-07-12 by FrancoisPicardDTI