basler/pylon-ros-camera #26: Dart camera timeout when using ROS (or Pylon)

jveitchmichaelis, 2020-03-01:
Camera: daA1600-60uc
Platform: Rock960A/B, ARM64
Pylon: 5.2.0

I'm sometimes unable to load the camera using ROS using the sample node - it doesn't open in Pylon either. It's not clear what the error is as the camera is detected:

```
[ WARN] [1583103522.287427099]: Autoflash: 0, line2: 1 , line3: 1
[ INFO] [1583103522.288132931]: No Device User ID set -> Will open the camera device found first
[ INFO] [1583103522.619771689]: Found camera with DeviceUserID N/A: daA1600-60uc
[ERROR] [1583103531.077220534]: Could not apply configuration. Pylon::GenericException caught in OnOpened method msg=Write operation failed on device '2676:ba03:3:1:2' at address: 0x10000038; length 4; Error: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsb
[ERROR] [1583103531.078270532]: Error while trying to open the desired camera!
```

When I try to run a simple capture script:

```
linaro@linaro-alip:~/tests$ cat test_basler.py
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

# demonstrate some feature access
new_width = camera.Width.GetValue() - camera.Width.GetInc()
if new_width >= camera.Width.GetMin():
    camera.Width.SetValue(new_width)

numberOfImagesToGrab = 100

camera.StartGrabbingMax(numberOfImagesToGrab)

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data.
        print("SizeX: ", grabResult.Width)
        print("SizeY: ", grabResult.Height)
        img = grabResult.Array
        print("Gray value of first pixel: ", img[0, 0])

    grabResult.Release()

camera.Close()
linaro@linaro-alip:~/tests$ python test_basler.py
Traceback (most recent call last):
  File "test_basler.py", line 4, in <module>
    camera.Open()
  File "/home/linaro/.local/lib/python2.7/site-packages/pypylon/pylon.py", line 3417, in Open
    return _pylon.InstantCamera_Open(self)
_genicam.RuntimeException: Could not apply configuration. Pylon::GenericException caught in OnOpened method msg=Read operation on device '2676:ba03:3:1:2' failed: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsbDevice.cpp', line 778) : RuntimeException thrown (file 'AcquireContinuousConfiguration.h', line 91)
```

This code definitely works, if I re-plug the camera:

```
linaro@linaro-alip:~/tests$ python test_basler.py
('SizeX: ', 1588)
('SizeY: ', 1200)
('Gray value of first pixel: ', 0)
('SizeX: ', 1588)
('SizeY: ', 1200)
('Gray value of first pixel: ', 0)
('SizeX: ', 1588)
('SizeY: ', 1200)
('Gray value of first pixel: ', 0)
('SizeX: ', 1588)
('SizeY: ', 1200)
('Gray value of first pixel: ', 0)
('SizeX: ', 1588)
('SizeY: ', 1200)
('Gray value of first pixel: ', 0)
```

Any idea what's going wrong? It's a bit concerning that whatever is failing seems to be locking up the USB device completely. I can provide more info if it would be useful.

pablo-quilez, 2020-04-09:
Hello,
I am not completely sure, but it seems to be related to the USB controller or the pylon library for ARM. Which operating system are you using? I would suggest you to test the camera with a x86 computer and e.g. Ubuntu to discard other possibilities.
Thanks!

m-binev, 2020-04-22:
Hi @jveitchmichaelis ,

I suppose you experience hardware issues. 
First of all, please replace the USB cable. I would recommend you to avoid using cheap USB cables, but recommended by Basler ones. The reported error (0x10000038) assumes that the connection to the camera is not stable, e.g. because of low-level errors.  
Also, please apply the following settings from the troubleshooting section:

Increasing Packet Size

For faster USB transfers you should increase the packet size. You can do this by changing the "Stream Parameters" -> "Maximum Transfer Size" value from inside the pylon Viewer or by setting the corresponding value via the API. After increasing the package size you will likely run out of kernel space and see corresponding error messages on the console. The default value set by the kernel is 16 MB. To set the value (in this example to 1000 MB) you can execute as root: echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb This would assign a maximum of 1000 MB to the USB stack.

pablo-quilez, 2020-07-20:
Hello, as no new infos where provided and ARM is not the main ROS architecture, I close the issue. Please don't hesitate to open the topic again if you have any question. Thanks!

closed 2020-07-20 by pablo-quilez

jveitchmichaelis, 2020-07-20:
Heya thanks - it's probably a custom cable issue. We can't use a normal USB3 cable as they're not flexible enough. I found some odd issues with port selection (the camera is quite picky about that) but it seems to be reliable in the configuration that I'm using now. I did also drop the bandwidth.

pablo-quilez, 2020-07-20:
Hi, nice infos. We hope that your set-up will continue working fine!