basler/pylon-ros-camera #31: The Image is dark , it looks like the green values are surpassing r and g values

jediofgever, 2020-04-01:
what is wrong with this camera , when i receive the image it has a weird format the looks darkish. 
I changed format to rgb8 but still cant geta normal looking rgb image, has anyone had same problem ?

jediofgever, 2020-04-01:
[image]

pablo-quilez, 2020-04-09:
Hello,

unfortunately as you are the only person having this or similar issues, we need more debugging information to proceed. Please provide:
- your camera model
- rest of configuration: gamma, brightness, etc.
- which format are you using for publishing the information

I suggest that you check directly with Pylon Viewer application to discard a hardware issue. https://www.baslerweb.com/en/products/software/basler-pylon-camera-software-suite/

Thanks!

closed 2020-04-09 by jediofgever