basler/pylon-ros-camera #32: ROS dependencies installation error

CloudYin, 2020-04-08:
Hello, 

when I am installing the ROS dependencies I got a following error. I have installed pylon 6.1.0 in /opt. Is there a way to solve it? 

reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Hit https://raw.githubusercontent.com/basler/pylon-ros-camera/master/pylon_camera/rosdep/pylon_sdk.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
updated cache in /home/pilz/.ros/rosdep/sources.cache
/opt/pylon/bin/pylon-config
Could not find any pylon Installation with version 5 or greater
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pylon_camera: Cannot locate rosdep definition for [dnb_msgs]

pablo-quilez, 2020-04-09:
Hello,

dnb_msgs is included here git clone https://github.com/dragandbot/dragandbot_common.git

You also need to clone it in your catkin_ws/src directory.

Regards

closed 2020-04-09 by pablo-quilez

CloudYin, 2020-04-10:
> Hello,
> 
> dnb_msgs is included here git clone https://github.com/dragandbot/dragandbot_common.git
> 
> You also need to clone it in your catkin_ws/src directory.
> 
> Regards

@pablo-quilez 
After I reinstall pylon 5.2.0 and dnb_msgs the rosdep is finished. However catkin_make is failed, some header files are missing.

[  0%] Building CXX object pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/src/pylon_camera/IPConfigAuto.cpp.o
[  0%] Building CXX object pylon-ros-camera/pylon_camera/CMakeFiles/write_device_user_id_to_camera.dir/src/pylon_camera/write_device_user_id_to_camera.cpp.o
/home/pilz/ROS_Projects/intellegent_platform_ws/src/pylon-ros-camera/pylon_camera/src/pylon_camera/IPConfigAuto.cpp:8:10: fatal error: pylon/PylonIncludes.h: No such file or directory
 #include <pylon/PylonIncludes.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/build.make:62: recipe for target 'pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/src/pylon_camera/IPConfigAuto.cpp.o' failed
make[2]: *** [pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/src/pylon_camera/IPConfigAuto.cpp.o] Error 1
CMakeFiles/Makefile2:3744: recipe for target 'pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/all' failed
make[1]: *** [pylon-ros-camera/pylon_camera/CMakeFiles/IPConfigAuto.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  0%] Built target diagnostic_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_py
/home/pilz/ROS_Projects/intellegent_platform_ws/src/pylon-ros-camera/pylon_camera/src/pylon_camera/write_device_user_id_to_camera.cpp:36:10: fatal error: pylon/PylonIncludes.h: No such file or directory
 #include <pylon/PylonIncludes.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

fmessmer, 2020-10-12:
@pablo-quilez 
could this dependency not be removed by moving the required package/message from `dragandbot_common` into the camera driver repo directly?

pablo-quilez, 2020-10-12:
Hi @fmessmer! 
We prefer not to do so because dragandbot_common is used by many other components and the camera driver is in-build with our deployment process in drag&bot. This means we would have a duplicated package. Maybe if you know a way to have duplicated packages in ROS following a priority would be ok.
Cheers!