basler/pylon-ros-camera #35: ROS Installation problem

RomanEVinnik, 2020-05-20:
Based on manual installation on stage - 
$ sudo sh -c 'echo "yaml https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_came ra/rosdep/pylon_sdk.yaml " > /etc/ros/rosdep/sources.list.d/30- plyon_camera.list' 
$ rosdep update

The ERROR received:

ERROR: unable to process source [https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_came]:
                HTTP Error 404: Not Found (https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_came)

RomanEVinnik, 2020-05-20:
ERROR: unable to process source [https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_camera/rosdep/pylon_sdk.yaml]:
	HTTP Error 404: Not Found (https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_camera/rosdep/pylon_sdk.yaml)

RomanEVinnik, 2020-05-24:
I think we fix it:
1. After installation of viewer need to run:  $ echo \u201cexport PYLON_ROOT=/opt/pylon\u201d >> ~/.bashrc instead of $ echo \u201cexport PYLON_ROOT=/opt/pylon5\u201d >> ~/.bashrc

2.  Run sudo sh -c 'echo "yaml https://raw.githubusercontent.com/basler/pylon-ros-camera/master/pylon_camera/rosdep/pylon_sdk.yaml " > /etc/ros/rosdep/sources.list.d/30-plyon_camera.list' instead of sudo sh -c 'echo "yaml https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_camera/rosdep/pylon_sdk.yaml " > /etc/ros/rosdep/sources.list.d/30-plyon_camera.list' 
like in Basler manual page 10.

pablo-quilez, 2020-05-28:
Hello Roman,

I will check it again. If I understand correctly, you installed first pylon manually and then you added the rosdep? The script behind will automatically install Pylon5 in the system. This may be the origin of the conflict. Anyway we will check it.

Cheers

pablo-quilez, 2020-05-28:
@RomanEVinnik I checked again and I saw that you cut the URL address here possibly by copy & paste. I think this was the main Issue. I will close this thread but please feel free to open again if you have any question.

```
$ sudo sh -c 'echo "yaml https://raw.githubusercontent.com/basler/pylon_ros_camera/master/pylon_came ra/rosdep/pylon_sdk.yaml " > /etc/ros/rosdep/sources.list.d/30- plyon_camera.list'
$ rosdep update
```

closed 2020-05-28 by pablo-quilez

AbdullahJirjees, 2024-09-10:
I am having issue with running 
~/catkin_ws$ roslaunch pylon_camera pylon_camera_node.launch
RLException: [pylon_camera_node.launch] is neither a launch file in package [pylon_camera] nor is [pylon_camera] a launch file name
The traceback for the exception was written to the log file

~/catkin_ws$ rosrun pylon_camera pylon_camera_node
[rospack] Error: package 'pylon_camera' not found