basler/pylon-ros-camera #36: "No camera present"after roslaunch pylon_camera pylon_camera_node.launch

Qin-xs, 2020-06-06:
Hi there,
 I use pylon on a linux arm system, but when I run "roslaunch pylon_camera pylon_camera_node.launch", "No camera present" appears, as follows:
process[pylon_camera_node-1]: started with pid [10841]
[ WARN] [1591426727.887867753]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1591426727.890951286]: No Device User ID set -> Will open the camera device found first
[ERROR] [1591426728.221549200]: No camera present
[ INFO] [1591426730.223653219]: using default calibration URL
[ INFO] [1591426730.224012859]: camera calibration URL: file:///home/nvidia/.ros/camera_info/camera.yaml
[ INFO] [1591426730.224407350]: Unable to open camera calibration file [/home/nvidia/.ros/camera_info/camera.yaml]
[ WARN] [1591426730.224656519]: Camera calibration file /home/nvidia/.ros/camera_info/camera.yaml not found.
[ WARN] [1591426744.554398780]: No camera present. Keep waiting ...

I can use the camera normally in the pylon viewer, but I cannot detect the camera in ros，what should i do?

pablo-quilez, 2020-07-24:
Hi,

which kind of camera are you using? I guess probably it is an Ethernet camera. Please try to assign a different IP address through the pylon IP Configurator program and check that your ip is in range. It may help.

Qin-xs, 2020-07-24:
Thank you for your reply.It works.

Qin-xs, 2020-07-24:
> 
> 
> Hi,
> 
> which kind of camera are you using? I guess probably it is an Ethernet camera. Please try to assign a different IP address through the pylon IP Configurator program and check that your ip is in range. It may help.

Thank you for your reply.It works.

pablo-quilez, 2020-07-24:
Thanks to you!

closed 2020-07-24 by pablo-quilez