basler/pylon-ros-camera #38: Timestamp

galardo86, 2020-06-21:
Hi,

Is it possible to get timestamp of grabbed image directly from the camera ?

BR,
Boris

pablo-quilez, 2020-07-20:
Hello,

currently is not possible. The timestamp included in the header, e.g. `seq: 54
stamp: 
  secs: 1595261417
  nsecs: 295772745
frame_id: "pylon_camera"
` is generated from ROS at the grabbing function
`img_raw_msg_.header.stamp = ros::Time::now(); `

If I am correct, currently there is no current interface for this. Also we cannot replace the timestamp in the header which needs to be the ROS timestamp. The sensor_msgs/Image is not including this. A possibility is to create some configurable parameter to decide between using the ROS timestamp or the camera timestamp. But ideally, we should keep always having both. The timestamp can be grabbed through the `uint64_t Pylon::GrabResult::GetTimeStamp()const` C++ Pylon function, but the format is different than the ROS format. What do you suggest? Where should this be placed? 

Thanks!

tim-fan, 2020-07-21:
Hello, just thought I'd mention this pull request to the Magazino repo appears to implement the desired behavior, at least for the case there PTP synchronization is enabled:
[https://github.com/magazino/pylon_camera/pull/38](https://github.com/magazino/pylon_camera/pull/38)
The conversion into ROS time is implemented here:
[https://github.com/magazino/pylon_camera/pull/38/files#diff-5ce449aaf9ffaf007cf068f7c3164c54R395-R401](https://github.com/magazino/pylon_camera/pull/38/files#diff-5ce449aaf9ffaf007cf068f7c3164c54R395-R401
)
A description of the implemented behaviour is provided:
```
- **fetch_camera_timestamp**
  Boolean set at start-up to choose which of {camera, host computer} is in charge of timestamping the image.
If the camera's clock is synchronized with the clock of the host computer (e.g. through PTP IEEE-1588), setting
this parameter to true improves the accuracy of the timestamps as this factors out network delays and IO delays
due to a high CPU load.
```

As I understand it, if PTP synchronization is not used, then the pylon timestamps will be reported in ticks since camera startup, at a frequency specific to the camera model, as documented here: [https://docs.baslerweb.com/timestamp#specifics](https://docs.baslerweb.com/timestamp#specifics ). In this case I suppose it would be possible to compute an offset and scale value to convert pylon timestamps to ROS timestamps based on the pylon timestamp of the first aquired image, but I would be wary of the camera's clock drifting from the host computer's clock.

Neel1302, 2020-08-27:
For anyone who is using the implementation in magazino/pylon_camera#38, please note the mapping of the camera ticks to time elapsed (see https://github.com/magazino/pylon_camera/pull/38#discussion_r478466074) for your camera model, until a fix is available.

FrancoisPicardDTI, 2022-02-07:
Closing this issue for now as there's no activity related to it for almost 2 years.

closed 2022-02-07 by FrancoisPicardDTI

tui-abdul, 2024-07-15:
Hi, 

Is this problem already solved, I am using ros2 Humble. I am trying to synchronize (PTP, IEEE 1588) Basler camera a2A2600-20gcPRO with the Lidar. Unfortunately the basler timestamps are coming from ros, which adds significant latency and mismatch in the lidar and camera frame. Basler having PTP functionality does not make sense if the timestamps are not coming from the camera itself.

tui-abdul, 2026-02-06:
> Hello, Has this problem been resolved? I also encountered the same situation. Can anyone offer some help? Thank you very much.

Yes, you have to enable chunk mode from pylon tool and select timestamps