basler/pylon-ros-camera #45: Read Pylon Temperature

tarkin2020, 2020-08-06:
Hello Dev-Team,

is it possible to add the psobility to read out the temperature of the camera?
Pylon sdk does support to read the temperature.

best regards

m-binev, 2020-08-06:
@tarkin2020 Hi, I'd like to politely point out that this is an open source project and you are highly welcome to add new camera parameter access yourselves. That is, there is no classical "Dev-Team", who is taking care of implementing new features for other users)) Regards

tarkin2020, 2020-08-06:
ok I would like to but have no idea how. Do you have any example?

m-binev, 2020-08-11:
@tarkin2020 Hi, you can consider the whole project as an example. Please check the implementation of similar parameters, e.g. Exposure Time, Gain, etc. in 'src/pylon_camera' and 'include/pylon_camera' and add new parameters accordingly. Regards

pablo-quilez, 2020-09-01:
Hi,

I have just added the temperature to the currentParams topic as new float field in the messsage. Internally uses DeviceTemperature for USB / Dart and TemperatureAbs for GigE cameras. There is a temperature/45 branch which I have merged into devel branch.

Cheers,
Pablo

Example of the topic:

```
rostopic echo /pylon_camera_node/currentParams 
offset_x: 8
offset_y: 4
reverse_x: False
reverse_y: False
black_level: 0
pgi_mode: -1
demosaicing_mode: -1
noise_reduction: -10000.0
sharpness_enhancement: -10000.0
light_source_preset: 1
balance_white_auto: -1
sensor_readout_mode: -1
acquisition_frame_count: 1
trigger_selector: 0
trigger_mode: 1
trigger_source: 0
trigger_activation: 0
trigger_delay: -10000.0
user_set_selector: 0
user_set_default_selector: 0
is_sleeping: False
brightness: 0.0
exposure: 3000.0
gain: 0.0
gamma: -1.0
binning_x: 1
binning_y: 1
roi: 
  x_offset: 8
  y_offset: 4
  height: 1080
  width: 1440
  do_rectify: False
available_image_encoding: [Mono8, BayerRG8, BayerRG12, BayerRG12Packed, YUV422Packed, YUV422_YUYV_Packed]
current_image_encoding: "Mono8"
current_image_ros_encoding: "mono8"
sucess: True
message: ''
temperature: 56.0
```

closed 2020-09-01 by pablo-quilez