basler/pylon-ros-camera #47: driver for two cameras

chentairan, 2020-08-14:
Hi,

I currently have two cameras, and I want to start them synchronously. What should I do?

Thanks,
Tairan Chen

ZQIONGZ, 2020-08-28:
hi，I also encountered the same problem. How did you solve it ?

m-binev, 2020-08-28:
@chentairan @ZQIONGZ Hi, why don't you check other issues in the backlog, e.g.:
https://github.com/basler/pylon-ros-camera/issues/34 
Other users seem to work fine with multiple cameras already. Of course, you have to adapt the code to your needs. Regards.

chentairan, 2020-08-28:
@m-binev Hi, do you have a time line for supporting the PTP and Muti-cameras features?

Thanks,
Tairan Chen

m-binev, 2020-08-28:
@chentairan Hi, this is an open source project and the source code is available for everyone to adapt and further develop it. That is, ther is no timeline for anything. Regarding PTP, there are at least two solutions available by other users that seem to work - so, please check them out., reuse and/or further develop them according to your needs. Cheers.

closed 2020-09-10 by chentairan