basler/pylon-ros-camera #52: Rosdep install uses incorrect architecture

tim-fan, 2020-09-14:
Hi there,

I'm trying to install pylon-ros-camera inside an i386 docker container. I'm building the container on an amd64 host.

Currently the `rosdep install` step fails because it tries to install `_pylon_5.2.0.13457-deb0_amd64.deb_` (i.e. the amd64 version of pylon).

Tracing this back, it looks like the amd64 version is picked based on the output of `uname -m` - see here:
https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/rosdep/pylon_sdk.rdmanifest#L16

Running `uname -m` inside the container returns `x86_64`, reflecting the fact that the container is running on a 64 bit host, but I need the 32 bit deb to be installed in this case.

As far as I can tell, a more suitable way to get the architecture for the package install would be `pkgarch=$(dpkg --print-architecture)`. I've confirmed that this returns `i386` in my container, and `amd64` on my host. I'm guessing that this approach would resolve my issue and allow you to get rid of the whole if/elif/else block here: 
https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/rosdep/pylon_sdk.rdmanifest#L16-L27

I could put this together as a pull request, but I'm not able to test on those other architectures, and I'm not sure if I'm missing something in my reasoning.

pablo-quilez, 2020-10-07:
Hi, 

Thanks for the input! I think `uname -m` is compatible with all kind of distributions while dpkg assumes debian-based, which may not be the case specially in the case of other architectures. It may not work with fedora, for example. Your case is special because you are running 32 bits container instead of running 64. Therefore I am not sure if changing to dpkg is the best solution.

As information, this rosdep was automated for drag&bot installation, originally you needed to download the pylon libraries from the main Basler site and install manually in the system from https://www.baslerweb.com/en/sales-support/downloads/software-downloads/

Cheers

tim-fan, 2020-10-10:
Ok, thanks for the comments - sounds like the current approach supports more platforms than a `dpkg` approach, so it wouldn't be worthwhile making the change for this specific use case (32 bit docker image). Will close the issue. 

> originally you needed to download the pylon libraries

Yes I much prefer the rosdep based method! Thanks again,
-Tim

closed 2020-10-10 by tim-fan

pablo-quilez, 2020-10-12:
@tim-fan thank to you for your interest! It was a nice point you discover here.