basler/pylon-ros-camera #53: Not finding camera with UserDefinedID

florisvb, 2020-09-20:
I tried setting my camera device_user_id using write_device_user_id_to_camera, which seems to work fine. However, when trying to run pylon_camera with the specified device_user_id, it only finds cameras with user_ids "N/A". 

My temporary solution was to replace every instance of "GetUserDefinedName" in pylon_camera with "GetDeviceGUID" and setting the device_user_id to be equal to the camera's GUID. This works great. 

I don't know enough c++ to debug why the original approach is not working :(

pablo-quilez, 2020-10-07:
Hello @florisvb, which camera model do you have? Which connection, USB or Ethernet? Are you using ACE2 or ACE model?

florisvb, 2020-10-07:
Basler ace, connected via USB3.

pablo-quilez, 2020-10-12:
Hi,

I tested today with a medical USB daA1600-60uc camera and it was working. Then I tried with a different USB camera, model acA2440 and you are right. DeviceUserID is not available. The `roslaunch pylon_camera pylon_camera_node.launch` will still work because takes the first available camera, but we can see:

```
[ INFO] [1602494514.497530781]: Found camera with DeviceUserID N/A: acA2440-35um
[ WARN] [1602494514.769154152]: No User Set Is selected, Camera current setting will be used
```

Through Pylon Viewer (downloadable from Basler website) I could see that the Device User ID under Device Control changed, but not the camera name in the list as it changed with the Dart camera. I will continue working on this topic and we will keep you posted about the issue.

Regards

pablo-quilez, 2020-10-12:
Hi,

I was speaking with Basler and got the solution. After rewriting the device user id, you need to reset the camera (take out the USB, plug in again). Then the correct name is displayed and can be used.

**Step 1**: change the name 

```
rosrun pylon_camera write_device_user_id_to_camera test
```
**Step 2**: reset the camera (usb reconnect)

**Step 3**: set the device_user_id

```
rosparam set /pylon_camera_node/device_user_id test
```
**Step 4**: start the camera node

```
rosrun pylon_camera pylon_camera_node 
[ WARN] [1602510267.135901258]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1602510267.135950574]: Trying to open the following camera: test
[ INFO] [1602510268.311538591]: Found the desired camera with DeviceUserID test: acA2440-35um
```

Regards,
Pablo

closed 2020-10-12 by pablo-quilez

florisvb, 2020-10-12:
Thanks for figuring that out!

- Floris