basler/pylon-ros-camera #58: version for ROS2-foxy

cf95, 2020-11-15:
Hi,

is there a similar implementation for ros2 foxy or any other ros2 version?

BR Chris

marccastor2, 2020-12-03:
Hi, I'm interested into your question, did you find any solution?

cf95, 2020-12-03:
Hi, I didn't find one yet, but i wasn't searching or developing one. I guess an easy solution would be to use the ros1 bridge (drawback: you would need a full ros1 installation on your device) or look into the code here, what has to be modified to get it working with ros2. Anyhow, depending on when you would need it, I have to look into this for my master thesis, but it may take a month or two until I get to the point, where I really need it. I can then report my solution here

pablo-quilez, 2020-12-07:
Hello,

I think makes a lot of sense to port the driver to ROS2. However, the necessary effort to do it is relative high. Therefore it is not a task which can start immediately. I keep open the issue for the future.

Cheers,
Pablo

bigrobinson, 2021-01-31:
I second this emotion. ROS2 drivers would be great.

MCP-TTC, 2021-02-03:
I agree, a ROS2 driver is a key component for ROS2 in industrial applications

jmarsik, 2021-07-07:
Is there any progress in this matter? We would also benefit from having a ROS2 native driver for Basler cameras.

fhwedel-hoe, 2021-07-07:
I could use some input regarding my driver, especially when it comes to performance. It does not support any kind of dynamic reconfiguration so it might be unsuitable for your needs.

jmarsik, 2021-07-07:
@fhwedel-hoe I think it would be ok. But our camera is connected through GigE Vision :(

jmarsik, 2021-07-13:
@fhwedel-hoe would it be complicated to add basic support for GigE Vision cameras to your driver?

fhwedel-hoe, 2021-07-14:
@jmarsik Code-wise, it is trivial. I replaced "Usb" with "GigE" and https://github.com/fhwedel-hoe/pylon_usb-instant-camera/tree/ros2_gige compiles. Since I do not have a Basler GigE camera at my hands, I cannot run tests so I have no idea whether this actually works. Feel free to give it a try (the package itself is still called pylon_usb_instant_camera with the PylonUSBCameraNode – I did not make the replacement in the names).