basler/pylon-ros-camera #60: Handle multiple USB cameras

gewi1011, 2020-12-08:
Hello,
How could i handle multiple USB Basler cameras?
I could successfully get the image of single camera. I set each camera a different UserDevice ID ("camera_01", "camera_02", etc.) and reset them afterwards. I changed the parameters in the default.yaml file at catkin_ws/src/pylon-ros-camera/pylon_camera/config as mentioned in the READ_ME File in the package folders: device_user_id: "camera_01" but it only detect one camera with its user ID. It tried also to add both camera-IDs (i tried something like: device_user_id: "camera_01", "camera_02") to the parameter but i got an error. I couldn't find any information how this driver work with multiple USB cameras. Is it necessary to run one node for each camera? And how do adapt some settings of a specific camera with rosservice, how do i adress the camera?
I appreciate any help

roboticslab-fr, 2020-12-08:
Hello @gewi1011 ,

Here is a simple solution I implemented to interface several BASLER cameras.

In a nutshell, to interface multiple cameras :
1 - create one yaml setting file by camera with the corresponding device_user_id
2 - create a launch file to launch one pylon_camera_node per camera using namespaces.

Implementation details are available on our forked repo : [https://github.com/roboticslab-fr/pylon-ros-camera](https://github.com/roboticslab-fr/pylon-ros-camera)

Example For 2 cameras named "ace_RGB_Left" & "ace_RGB_Right" :

YAML files :
[ace_RGB_Left.yaml](https://github.com/roboticslab-fr/pylon-ros-camera/blob/master/pylon_camera/config/ace_RGB_Left.yaml)
[ace_RGB_Right.yaml](https://github.com/roboticslab-fr/pylon-ros-camera/blob/master/pylon_camera/config/ace_RGB_Right.yaml)

Launch file :
[pylon_camera_node_2RGB.launch](https://github.com/roboticslab-fr/pylon-ros-camera/blob/master/pylon_camera/launch/pylon_camera_node_2RGB.launch)

Images are then published to :
```
/ace_RGB_Left/pylon_camera_node/image_raw
/ace_RGB_Right/pylon_camera_node/image_raw
```

We are using this implementation with 4 USB Basler cameras (and also for 2 GigE cameras).
I hope it will be helpful.

gewi1011, 2020-12-10:
Thanks for your help! Everything is working now!

pablo-quilez, 2020-12-11:
Thank you for the answer and details @roboticslab-fr !
I close the ticket.

closed 2020-12-11 by pablo-quilez