basler/pylon-ros-camera #62: pylon/PylonIncludes.h: No such file or directory

RGring, 2020-12-11:
Hello!
I tried to install the package on a Jetson device (ARM) and during catkin_make, I get the following error:
```
fatal error: pylon/PylonIncludes.h: No such file or directory
 #include <pylon/PylonIncludes.h>
```

Pylon is installed at /opt/pylon5 as expected (and stated in the FindPylon.cmake). 
Do you have any ideas, how I can get it working?

pablo-quilez, 2020-12-11:
Hello, I suggest to check the PYLON_ROOT environment variable from this file https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/cmake/FindPylon.cmake.

RGring, 2020-12-11:
Yes it is set to the correct path: /opt/pylon5.

RGring, 2020-12-11:
By installing pylon via https://www.baslerweb.com/en/sales-support/downloads/software-downloads/#type=pylonsoftware;version=all, it worked. Thanks anyway, I close the issue.

ChuangLi32782019, 2021-05-14:
when you installed the pylonsoftware, it works?

raabuchanan, 2022-09-20:
I had a similar issue because I didn't realize pylon had to be separately installed. I suggest to the maintainers to add a link to the download page for `pylon6` to the ReadMe so others don't get stuck like me.