basler/pylon-ros-camera #63: Error while start grabbing. ROS melodic, pylon 6, Basler Ace 2 Ethernet camera.

mpQuintana, 2020-12-18:
Hello, 

I have a Basler ace 2 camera (a2A1920-51gcBAS) connected directly to my laptop with static IPs configured (169.254.1.10 laptop, 169.254.1.1 camera). The pylon viewer works as expected and I have also tried some of the samples of the SDK and they work as expected but I can not get the ROS node to work. It throws the error "Error while start grabbing". I installed everything following the readme but I had pylon 6 installed.

This is the output of roslaunch pylon_camera pylon_camera_node.launch with de configuration set to debug=true. Any idea on what might be the issue and how to solve it?

marc@pc2829pt:~/basler_catkin_ws$ roslaunch pylon_camera pylon_camera_node.launch 
... logging to /home/marc/.ros/log/dd105d88-4130-11eb-8d7c-1065303dcaa8/roslaunch-pc2829pt-26288.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pc2829pt:33101/

SUMMARY
========

PARAMETERS
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: 
 * /pylon_camera_node/enable_current_params_publisher: True
 * /pylon_camera_node/enable_status_publisher: True
 * /pylon_camera_node/frame_rate: 15.0
 * /pylon_camera_node/gige/inter_pkg_delay: 1000
 * /pylon_camera_node/gige/mtu_size: 1500
 * /pylon_camera_node/image_encoding: bayer_gbrg8
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    pylon_camera_node (pylon_camera/pylon_camera_node)

ROS_MASTER_URI=http://localhost:11311

process[pylon_camera_node-1]: started with pid [26306]
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
---Type <return> to continue, or q <return> to quit---
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Registered pretty printers for UE4 classes
Reading symbols from /home/idiada/Ensemble/basler_catkin_ws/devel/lib/pylon_camera/pylon_camera_node...(no debugging symbols found)...done.
Starting program: /home/marc/basler_catkin_ws/devel/lib/pylon_camera/pylon_camera_node __name:=pylon_camera_node __log:=/home/idiada/.ros/log/dd105d88-4130-11eb-8d7c-1065303dcaa8/pylon_camera_node-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffe67a4700 (LWP 26313)]
[New Thread 0x7fffe5fa3700 (LWP 26314)]
[New Thread 0x7fffe57a2700 (LWP 26315)]
[New Thread 0x7fffe4fa1700 (LWP 26320)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[... 86 lines cut ...]
          inet6 addr: fe80::f354:5384:b003:6a4e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3304788 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19185 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4963976382 (4.9 GB)  TX bytes:1196259 (1.1 MB)
          Interrupt:16 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:54379 errors:0 dropped:0 overruns:0 frame:0
          TX packets:54379 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:13961275 (13.9 MB)  TX bytes:13961275 (13.9 MB)

wlp62s0   Link encap:Ethernet  HWaddr 9c:b6:d0:6b:70:a3  
          inet addr:192.168.248.68  Bcast:192.168.255.255  Mask:255.255.248.0
          inet6 addr: fe80::8180:fdc5:e184:2df5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:722113 errors:0 dropped:0 overruns:0 frame:0
          TX packets:415795 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:690722733 (690.7 MB)  TX bytes:154309170 (154.3 MB)

m-binev, 2020-12-23:
Hi @mpQuintana , the pylon ROS driver is based on pylon 5.2. As pylon 6.x and pylon 5.2 are not binary compatible, I would suggest that you install pylon 5.2 instead. Another thing to consider is, that the current ROS adapter was designed for GigE cameras with GenICam SFNC v1.1 and new Basler ace2 GigE camera (like yours) are based on GenICam SFNC v2.x. This means that the ace2 camera does use partly different names for some camera features. In other words, some modifications to the ROS adapter may be needed in order to get that camera with all its features working. 
@pablo-quilez FYI

mpQuintana, 2021-01-04:
Hello @m-binev, thank you for your answer. I have tried with pylon 5.2 but now I get a different error: "what():  Error while accessing GainRaw in PylonGigECamera". I am trying to look into the implications of using GenICam SFNC v2.x instead of v1.1, this means I should modify the PylonCameraParameter? Or what would I need to modify? 

This is the console output:

started roslaunch server http://pc2829pt:36537/

SUMMARY
========

PARAMETERS
 * /pylon_camera_node/brightness: 100
 * /pylon_camera_node/brightness_continuous: True
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: 
 * /pylon_camera_node/enable_current_params_publisher: True
 * /pylon_camera_node/enable_status_publisher: True
 * /pylon_camera_node/frame_rate: 5.0
 * /pylon_camera_node/gain: 0.5
 * /pylon_camera_node/gain_auto: True
 * /pylon_camera_node/gige/inter_pkg_delay: 1000
 * /pylon_camera_node/gige/mtu_size: 1500
 * /pylon_camera_node/image_encoding: rgb8
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    pylon_camera_node (pylon_camera/pylon_camera_node)

auto-starting new master
process[master]: started with pid [1566]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 242967ca-4e72-11eb-be7b-1065303dcaa8
process[rosout-1]: started with pid [1579]
started core service [/rosout]
process[pylon_camera_node-2]: started with pid [1591]
gain is given and has value 0.5
brightness is given and has value 100
brightness is continuous
gain is set to auto
[ WARN] [1609753771.139300776]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1609753771.139345974]: No Device User ID set -> Will open the camera device found first
[ INFO] [1609753771.395447129]: Found camera with DeviceUserID camera_1: a2A1920-51gcBAS
[ WARN] [1609753771.623949934]: No User Set Is selected, Camera current setting will be used
[ INFO] [1609753771.654818164]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] 
[ INFO] [1609753771.828643476]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
[ WARN] [1609753771.828699588]: Will only provide distorted /image_raw images!
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error while accessing GainRaw in PylonGigECamera
[pylon_camera_node-2] process has died [pid 1591, exit code -6, cmd /home/idiada/Ensemble/basler_catkin_ws/devel/lib/pylon_camera/pylon_camera_node __name:=pylon_camera_node __log:=/home/idiada/.ros/log/242967ca-4e72-11eb-be7b-1065303dcaa8/pylon_camera_node-2.log].
log file: /home/idiada/.ros/log/242967ca-4e72-11eb-be7b-1065303dcaa8/pylon_camera_node-2*.log

m-binev, 2021-01-04:
@mpQuintana Hi, well, this is the "correct" error message you could expect)) That is, the ace2 does not have any GainRaw (int), ExposureTimeRaw (int), etc. Instead, it has Gain (float), ExposureTime (float), etc. Thus, it would be better if you add these new parameters to the GigE camera implementation. 
FYI: the ace2 GigE cameras can be treated like (compared to) ace USB cameras, because both comply with SFNC 2.x. 
In other words, you can have a look at similar parameter implementation of the USB camera in ROS and replicate it for GigE with SFNC 2.x. Regards

mpQuintana, 2021-01-04:
Thank you @m-binev that is very helpful. I am trying to follow your suggestion but I am getting errors when compiling.
I am modifying the file pylon_camera/include/internal/impl/pylon_camera_gige.hpp to change from GainRaw to Gain for example (I have tried different attributes), Do I need to modify this in any other files? I think I'm missing something.

This is the code modified:
[image]

And this is the error:
[image]

m-binev, 2021-01-05:
@mpQuintana Hi, okay, I'm not a ROS developer, but I suppose I know where the problem is - let me elaborate on that.
1) I suppose, you have tried to modify the access to GainRaw (int, sfnc 1.1) to Gain (float, sfnc2.x). I suppose, you have also modified the typedef in the beginning of the same file.
2) based on the error message, e.g. "class Pylon::CBaslerGigEInstantCamera has no member named Gain", I assume that the current ROS port for GigE cameras is based on the native pylon class CBaslerGigEInstantCamera. 
Native means, that all camera parameters are members of that class and have been already generated based on camera parameter XML files that describe a SFNC 1.x GigE camera in pylon.
3) in turn, it could be very tricky and complicated to add support for a completely new parameter while using a native pylon class that does not know anything about that.
4) for such cases in pylon there is the so-called generic class, i.e., CInsantCamera. 
With this class you first pick up the camera nodemap directly from the camera (it describes all its parameters), and may then access any camera parameter. This approach is described in the pylon C++ SDK sample "ParametrizeCamera_GenericParameterAccess". Thus, I suppose this would be the right way to go here, if ace2 GigE should be supported in the current ROS adapter with pylon 5.2.
By the way, pylon Viewer is using exactly this generic class/approach, which is why it works well with ace2 GigE. 

Summary: 
I suppose a little more implementation would be needed in order to get the ace2 GigE working with the current ROS adapter. ace2 USB should be working out of the box though.
If pylon 6.x should be used with ROS (because it natively supports ace2 GigE), the new class CBaslerUniversalInstantCamera should be used instead of CBaslerGigEInstantCamera. 

@pablo-quilez FYI.

nickzherdev, 2021-01-09:
Hi!
Luckily, I saw this thread before trying to run my Baslers 1920-51gcBAS with pylon 6.1.1.
May I ask, when the support for that camera type be added natively into the driver?  Approximately?

m-binev, 2021-01-11:
@nickzherdev Hi, we are currently discussing on that and hope to get some decision until end of the month. If the decision was positive, we would need about 1 month to get things implemented. So, some time in March (or April) could do.
@pablo-quilez FYI.

mpQuintana, 2021-03-22:
Hello @m-binev. Any news regarding the support of ACE 2 GigE Cameras in ROS? We have ACE 2 BASIC GigE cameras and will have ACE 2 PRO GigE cameras soon, and we need to integrate them with ROS soon.

m-binev, 2021-03-22:
@mpQuintana Hi, we are working on that already and hope to have a new brach based on pylon 6.2 that supports ace 2 GigE cameras by e.g. middle of April. 
@pablo-quilez FYI.

pablo-quilez, 2021-04-21:
Hello,

we updated on last Friday the [pylon6-dev](https://github.com/basler/pylon-ros-camera/tree/pylon6-dev) branch. This should solve the issue, as we upgrade the driver for the missing cameras. Please let me know if it is working properly.

Thanks and regards!

FrancoisPicardDTI, 2022-02-07:
I am closing this issue as it seems that the problem has been solved.

closed 2022-02-07 by FrancoisPicardDTI