basler/pylon-ros-camera #64: Increasing bandwith ROS Topic Image_raw

hannespimmel, 2020-12-27:
Hello, i am a beginner with the ros pylon package and i want to ask how to solve the following problem:

I am trying to increase the bandwith and so on my publishing rate to the topic "image_raw".
I already changed the USB memory from Ubuntu as mentioned in the instuctions of the ros pylon package on github. I also tried to set the max usb transfer size to 150MB by rosservice commands but i didn't get any changes to my bandwith.

I am meassuring the bandwith by the ros call: "rostopic bw "..../image_raw" (for publishing rate: hz instead of bw).
At the moment i reach with one camera: publishing rate:  ca. 6,2 Hz and bandwidth: ca 12MB/s 

Could anybody help me with this issue?
In the end i want to work with several cameras at the same time so i should solve this problem....

Thanks for any help!

FranekStark, 2020-12-27:
Hi,
I'm not able to solve your problem but I had the same problems like you some time ago. 
In my opinion this node is a bit 'messy' which means it isn't behaving like I would expect and it isn't implemented like ROS would recommend.
Here are some adviced that helped me:
- Make sure that you use a USB3.0 connection I know that the cameras are working with USB2.0 but it isn't recommended.
- Play a bit with the exposure time. I'm not really sure how to explain it, but sometimes it helps.
- Try to watch the compressed images. 
- Simply reduce the resolution. I know it's stupid, but this was finally my way to go, as I wasn't get acceptable frame rates out of it.
- Are you watching the images on the device on which the camera is connected to? Otherwise the network transport could be also a big bottle neck.

hannespimmel, 2020-12-27:
Thanks for the quick answer!
I think the problem is the ROS environment, because when i connect the camera with the same settings(bayer_rgb, resolution etc.) to the pylon viewer, i reached 40 fps and something round about 160 MB/s. 
I changed the max USBFS MEMORY LIMIT in ubuntu to 1000MB, so ubuntu should be also not the problem.
So it should be possible to reach more than 6 fps and 12 MB/s for the image_raw topic...

FranekStark, 2020-12-28:
Yeah I would agree. I think that the Ros Node is somehow a limiting factor. 
Are you trying to vizualize the images with Rviz or rqt? Maybe they are also a problem? Have you tried to receive the images in your own node? What data rates are you reaching?

FranekStark, 2020-12-28:
It seems that the developers have fixed some issues in the meatime since I've had this issues:
https://github.com/basler/pylon-ros-camera/issues/29
https://github.com/basler/pylon-ros-camera/issues/21
https://github.com/basler/pylon-ros-camera/issues/28

Unfortunately, I have no camera over here to verify that but I think it will solve your problems!

m-binev, 2020-12-28:
@hannespimmel @FranekStark Hi, if you excluded any limitations due to exposure time and the camera being operatzed in USB2.0 mode, then it is because of the Software Trigger mode, which is by default used in the ROS adapter implementation. This mode naturally limits the maximum possible frame rate by at least a half. Thus, check especially #28 and use either free-run (like default in pylon viewer) or hardware trigger (you'll need an external cabling and trigger then). Cheers