basler/pylon-ros-camera #66: Error connecting to Basler ACE 2 Pro

tobias-fleck-fzi, 2021-01-15:
Hello, 

I have issues running a basler ace 2 pro (a2A1920-51gcPRO) camera with the driver. I'm using ros noetic on 20.04. 
The error log shows that the  property ExposureTimeAbs cannot be queried from the camera. 

> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /pylon_camera_node/camera_frame: pylon_camera
>  * /pylon_camera_node/camera_info_url: 
>  * /pylon_camera_node/device_user_id: 
>  * /pylon_camera_node/enable_current_params_publisher: True
>  * /pylon_camera_node/enable_status_publisher: True
>  * /pylon_camera_node/frame_rate: 5.0
>  * /pylon_camera_node/gige/mtu_size: 1500
>  * /pylon_camera_node/startup_user_set: CurrentSetting
>  * /rosdistro: noetic
>  * /rosversion: 1.15.9
> 
> NODES
>   /
>     pylon_camera_node (pylon_camera/pylon_camera_node)
> 
> auto-starting new master
> process[master]: started with pid [50947]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 56dfbef4-574a-11eb-af70-df158f63f6b0
> process[rosout-1]: started with pid [50957]
> started core service [/rosout]
> process[pylon_camera_node-2]: started with pid [50964]
> [ WARN][/pylon_camera_node][WallTime: 1610726236.868098176]: Autoflash: 0, line2: 1 , line3: 1 
> [ INFO][/pylon_camera_node][WallTime: 1610726236.868819414]: No Device User ID set -> Will open the camera device found first
> [ INFO][/pylon_camera_node][WallTime: 1610726237.124932171]: Found camera with DeviceUserID 0: a2A1920-51gcPRO
> [ WARN][/pylon_camera_node][WallTime: 1610726237.346371877]: No User Set Is selected, Camera current setting will be used
> [ INFO][/pylon_camera_node][WallTime: 1610726237.373381813]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] 
> [ WARN][/pylon_camera_node][WallTime: 1610726237.373513115]: No image encoding provided. Will use 'mono8' or 'rgb8' as fallback!
> [ INFO][/pylon_camera_node][WallTime: 1610726237.515286564]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
> [ WARN][/pylon_camera_node][WallTime: 1610726237.515392230]: Will only provide distorted /image_raw images!
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Error while accessing ExposureTimeAbs in PylonGigECamera

When I use the ./PylonViewerApp, I can access the camera and even view a video stream, so the cam in general seems to be working.
After I tried the ros driver, the camera seems to freeze internally and does not show any video stream in ./PylonViewerApp anymore (ros nodes were already terminated properly). 

Do you have any suggestions how to trace the problem? What did I do wrong?

thanks in advance for your help, 
Tobias

m-binev, 2021-01-15:
@tobias-fleck-fzi Hi, please check #63. Your camera model is not yet supported by the current ROS driver. We are considering to update it on basis of pylon 6.x in order to support ace 2 GigE either (ace 2 USB work fine). I suppose, this will take some time till March/April ,21. Regards.

tobias-fleck-fzi, 2021-01-15:
Thank you for your instant reply! Of course this explains the error. March/April sounds good for me. Is it possible to keep me updated regarding the ros support for ACE 2 pro, at least when where will be any delays? 
Best Wishes, Tobias

pablo-quilez, 2021-01-22:
Hello @tobias-fleck-fzi,
we will keep you posted, I keep the issue open.
Thanks!

michael1309, 2021-04-01:
Hello all,

What is the current status for this issue?
We want to use Ace2 cameras with ros and need a driver for it.
We had the same errors as @tobias-fleck-fzi 
We were able to patch the current driver so that at least images are delivered. But we can't configure the camera and especially the exposure time.

Tanks in Advance

m-binev, 2021-04-01:
@michael1309 Hi, we are still working on that and hope to have some working implementation until end of April. Regards.
@pablo-quilez FYI.

pablo-quilez, 2021-04-21:
Hello,

we updated on last Friday the [pylon6-dev](https://github.com/basler/pylon-ros-camera/tree/pylon6-dev) branch. This should solve the issue, as we upgrade the driver for the missing cameras. Please let me know if it is working properly.

Thanks and regards!

tfleck90, 2021-04-21:
Hello, 

I tested the new branch on ubuntu 20.04, pylon 6.2, with the mentioned ace 2 pro model above. The used commit was efc99321bb351c6d22651f13489b25e6e386b887. 

I was able to receive camera images from the driver and the driver seems to properly use all parameters I have given via config file so far.

Thanks for your awesome work!

Best wishes, 
Tobias

tobias-fleck-fzi, 2021-04-21:
Hello,

I tested the new branch on ubuntu 20.04, pylon 6.2, with the mentioned ace 2 pro model above. The used commit was efc9932.

I was able to receive camera images from the driver and the driver seems to properly use all parameters I have given via config file so far.

Thanks for your awesome work!

Best wishes,
Tobias

brinij, 2021-05-31:
Hello, 

I am trying to run ace 2 Pro Basler camera with the pylon6-dev branch on ubuntu 20.04.
It was working fine until I had to configure the flashlight I/O parameters.
I want to use the startup_user_set parameter set to  "Default" by the default, instead of "CurrentSetting", but when I do that camera fails to launch properly. It throws this error : `Error applying cam specific startup setting for GigE cameras: Parameter not found in CFloatParameter::GetMax. (No node attached.)`. Do you know which parameter is a problematic one here?

closed 2022-05-18 by FrancoisPicardDTI

leiwanrobotics, 2024-11-27:
> Hello,
>
> I am trying to run ace 2 Pro Basler camera with the pylon6-dev branch on ubuntu 20.04. It was working fine until I had to configure the flashlight I/O parameters. I want to use the [...]

Same issue