basler/pylon-ros-camera #67: Error while catkin_make

Kanghongsheng, 2021-01-24:
An error occurs after catkin_make command:

[ 98%] Building CXX object pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/src/pylon_camera/main.cpp.o
[100%] Linking CXX executable /home/khs/桌面/baslerCamera/pylon_ros/devel/lib/pylon_camera/pylon_camera_node
/home/khs/桌面/baslerCamera/pylon_ros/devel/lib/libpylon_camera.so：对‘cv::Mat::Mat()’未定义的引用
collect2: error: ld returned 1 exit status
pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/build.make:170: recipe for target '/home/khs/桌面/baslerCamera/pylon_ros/devel/lib/pylon_camera/pylon_camera_node' failed
make[2]: *** [/home/khs/桌面/baslerCamera/pylon_ros/devel/lib/pylon_camera/pylon_camera_node] Error 1
CMakeFiles/Makefile2:4048: recipe for target 'pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/all' failed
make[1]: *** [pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

before this step, everything goes right, how to solve this problem? thanks~

Kanghongsheng, 2021-01-25:
/home/khs/ros_workspace/devel/lib/libpylon_camera.so: undefined reference to `cv::Mat::Mat()'
collect2: error: ld returned 1 exit status
pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/build.make:170: recipe for target '/home/khs/ros_workspace/devel/lib/pylon_camera/pylon_camera_node' failed
make[2]: *** [/home/khs/ros_workspace/devel/lib/pylon_camera/pylon_camera_node] Error 1
CMakeFiles/Makefile2:4048: recipe for target 'pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/all' failed
make[1]: *** [pylon-ros-camera/pylon_camera/CMakeFiles/pylon_camera_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

pablo-quilez, 2021-02-04:
Hello,

I think some dependencies are missing in your system, probably cv_bridge. Did you call rosdep? 

`rosdep update && sudo rosdep install --from-paths . --ignore-src --rosdistro=$ROS_DISTRO -y`

Regards

closed 2021-03-05 by pablo-quilez