basler/pylon-ros-camera #69: Unable to grab images from daA3840-45uc

e-team-t, 2021-02-05:
`roslaunch pylon_camera pylon_camera_node.launch`   returns the following error, bit works fine with a daA2500-14uc

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dev.local:39093/

SUMMARY
========

PARAMETERS
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: 
 * /pylon_camera_node/enable_current_params_publisher: True
 * /pylon_camera_node/enable_status_publisher: True
 * /pylon_camera_node/frame_rate: 5.0
 * /pylon_camera_node/gige/mtu_size: 1500
 * /pylon_camera_node/image_encoding: rgb8
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    pylon_camera_node (pylon_camera/pylon_camera_node)

auto-starting new master
process[master]: started with pid [21582]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3ebc001e-67d3-11eb-a847-00044bec1ccd
process[rosout-1]: started with pid [21593]
started core service [/rosout]
process[pylon_camera_node-2]: started with pid [21597]
[ WARN] [1612544257.249715311]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1612544257.251923600]: No Device User ID set -> Will open the camera device found first
[ INFO] [1612544257.525473588]: Found camera with DeviceUserID N/A: daA3840-45uc
[ WARN] [1612544257.934902505]: No User Set Is selected, Camera current setting will be used
[ INFO] [1612544257.948000936]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT'] ['RGB8'|'rgb8'] ['BGR8'|'bgr8'] ['YCbCr422_8'|'NO_ROS_EQUIVALENT'] ['BayerRG8'|'bayer_rggb8'] ['BayerRG12'|'bayer_rggb16'] ['BayerRG12p'|'NO_ROS_EQUIVALENT'] 
[ERROR] [1612544259.000065048]: Waiting for Trigger signal
[ERROR] [1612544259.000201458]: PylonCamera not ready because the result of the initial grab is invalid
[ WARN] [1612544259.018461124]: [] name not valid for camera_info_manager
[ INFO] [1612544259.038553393]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
[ WARN] [1612544259.038646625]: Will only provide distorted /image_raw images!
[ INFO] [1612544259.040987678]: Startup settings: encoding = 'rgb8', binning = [1, 1], exposure = 14983, gain = 0, gamma = 1, shutter mode = default_shutter_mode
[ INFO] [1612544259.041788794]: Desired framerate 5.00 is higher than max possible. Will limit framerate to: 0.30 Hz
[ INFO] [1612544259.044071617]: Start image grabbing if node connects to topic with a frame_rate of: 0.297531 Hz
[ INFO] [1612544259.044281518]: Camera not calibrated
[ WARN] [1612544259.045840103]: Error in triggerDeviceReset(): pylon_camera_ is not ready!
[ WARN] [1612544262.406677254]: Error in triggerDeviceReset(): pylon_camera_ is not ready!
[ WARN] [1612544265.774585948]: Error in triggerDeviceReset(): pylon_camera_ is not ready!
```

m-binev, 2021-02-15:
@e-team-t Hello. Your camera has a prototype status yet and it has not been tested with the ROS adapter. We are going to get a series camera in the beginning/middle of March first, and hope to fix any potential issues in April/May at earliest.

pablo-quilez, 2021-03-16:
Hello, we have released the following branch https://github.com/basler/pylon-ros-camera/tree/pylon6-dev. Your camera should work with it (USB cameras are already integrated, GigE is next step).

closed 2021-06-14 by pablo-quilez