basler/pylon-ros-camera #70: Unable to run the pylon_camera_node with basler blaze camera

robograffitti, 2021-02-11:
Now, I'm following the manual here:
https://www.baslerweb.com/fp-1591878203/media/downloads/documents/application_notes/AW00149103000_Interfacing_Basler_Cameras_with_ROS~1.pdf

`$ rosrun pylon_camera pylon_camera_node `

returned the following error:

```
[ WARN] [1613029125.159595628]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1613029125.160155182]: No Device User ID set -> Will open the camera device found first
[ INFO] [1613029125.467144068]: Found camera with DeviceUserID : blaze-101-GEV
[ INFO] [1613029125.837165899]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono16'|'mono16'] 
[ WARN] [1613029125.837357976]: No image encoding provided. Will use 'mono8' or 'rgb8' as fallback!
[ERROR] [1613029125.837498203]: Couldn't find a fallback solution!
[ERROR] [1613029125.837626693]: Error while start grabbing

```

Then, I tried:
`$ rosparam set /pylon_camera_node/image_encoding "mono16"`

still returned the error:

```
$ rosrun pylon_camera pylon_camera_node 
[ WARN] [1613029547.989726418]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1613029547.990249293]: No Device User ID set -> Will open the camera device found first
[ INFO] [1613029548.299452267]: Found camera with DeviceUserID : blaze-101-GEV
[ INFO] [1613029548.663975655]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono16'|'mono16'] 
[ERROR] [1613029548.664156572]: Error while start grabbing
```

m-binev, 2021-02-11:
@robograffitti Hi, this is the wrong project for you(( This pylon ROS adapter and the AppNotes are ony for Basler GigE & USB3 cameras and not for 3d/TOF blaze. For using the blaze camera on ROS, please download the blaze SDK from the Basler website and go to the local ROS folder for more information and guidance. Regards

pablo-quilez, 2021-03-05:
Hello @robograffitti, this repository is the false driver. The correct one can be found here: https://www.baslerweb.com/en/sales-support/downloads/document-downloads/interfacing-basler-cameras-with-ros/

Regards

closed 2021-03-05 by pablo-quilez