basler/pylon-ros-camera #71: Unable to run pylon_camera_node with BASLER scA1390-ABB camera

SaengthongStk, 2021-03-04:
I'm using BASLER scA1390-ABB camera with pylon_camera package and ROS-kinetic. 
I can't connect to the camera with automatic IP address, So I configured IPv4 Setting in Manual, Address: 169.254.1.1 and  Netmask: 255.255.0.0  Finally the connection is established.
Then I tried the following command:

`roslaunch pylon_camera pylon_camera_node.launch`

And then it returned the following error:

```
[ WARN] [1614848016.079406596]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1614848016.079532262]: No Device User ID set -> Will open the camera device found first
[ERROR] [1614848016.386277604]: No camera present
```
I'm not sure can I use the camera with pylon_camera package.
What am I doing wrong?

m-binev, 2021-03-04:
@SaengthongStk Hi, I have two assumptions for your issue:
1) you are using an OEM (customized) camera that may be treated as non-Basler camera by pylon, which means that pylon may reject its usage. 
In such cases, OEMs typically get a pylon license key to operate such cameras in pylon though. To check and confirm that, please open the camera in pylon Viewer (if you can at all - see 2), in the Features pane open the Device Information node and make a screenshot of it - then post it here or get back to the Basler support.
2) your IP configuration is wrong (also because of having multiple NICs with Auto IP in the system). To verify that, can you open the pylon Viewer and/or the IP Configuration Tool and check you find and can run the camera?
Alternatively, please configure your NIC and camera to static IP adresses in Class C, e.g.,:
NIC: 192.168.22.2, Subnet Mask= 255.255.255.0
cam: 192.168.22.5, Subnet Mask= 255.255.255.0

As a side note: your camera is relatively old and already discontinued. See here:
https://www.baslerweb.com/en/sales-support/after-sales-services/discontinued-products/discontinued-cameras/

SaengthongStk, 2021-03-05:
I don't know how can I open  the Device Information node ,But I try to open the camera in pylon viewer both on Window and Ubuntu. The following  picture is pylon viewer from the Window.

[image]

I can open the camera once with pylon viewer on Ubuntu but it was suddenly disconnected and I tried to reconnect without success. The following picture is pylon viewer from the Ubuntu.

[image]

m-binev, 2021-03-05:
@SaengthongStk Hi, I think it is not an issue related to an OEM camera, because you obviously can open and run the camera in pylon viewer.
However, on one of the screenshots I saw that you are using pylon 5.0.12, which is too old for this ROS driver. That is, the current ROS driver is using pylon 5.2.0. So please make sure to install and use 5.2.0 instead.
Under this link you can go to the downloads and check the ROS AppNotes:
https://www.baslerweb.com/en/products/software/basler-pylon-camera-software-suite/pylon-open-source-projects/ 
https://www.baslerweb.com/fp-1591878203/media/downloads/documents/application_notes/AW00149103000_Interfacing_Basler_Cameras_with_ROS~1.pdf

pablo-quilez, 2021-03-05:
Hello @SaengthongStk,

Please check that your pc ip address is also in the last subrange ignoring the fact that the mask is 255.255.0.0. This means in your case that:
- Camera ip: 169.254.1.1
- Netmask: 255.255.0.0
- PC ip: 169.254.1.x
If this is not working, you may try to change the mask to 255.255.255.0

Please let me know if it works, maybe is useful information to track in the README.

Regards,
Pablo

SaengthongStk, 2021-03-05:
@m-binev Hi, I installed pylon 6.1.1 Camera Software Suite Linux x86 (64 Bit) (on my Ubuntu 16.04 PC), Then open pylon viewer to connect the camera ,But it still show "No Devices found".

SaengthongStk, 2021-03-05:
@pablo-quilez Hi, I'm not sure that I configure PC IP address correctly. The following picture is IP address configuration.

[image]

I saved the configuration, But it is still show "No Devices found" anyway.

m-binev, 2021-03-05:
@SaengthongStk Hi, let's start with some basics about IP addresses and subnet masks)).
Please do the following:
1) from the terminal: 
sudo ifconfig enp27s0 192.168.1.2
the subnet mask muss be automitacially set to: 255.255.255.0. Please check that by calling ifconfig in the terminal.
2) open pylon Viewer > tools > IpConfigurator > Static IP > set the camera to:
192.168.1.5, 255.255.255.0.
3) try to run the camera in pylon viewer, i.e. grab images continuously.
If that works fine, you could try ROS again.

SaengthongStk, 2021-03-08:
@m-binev Hi, I did the first and second step ,But it still can't find any camera as shown in the following picture.

[image]

m-binev, 2021-03-08:
@SaengthongStk Okay, you have to first make sure the camera is usable in pylon.
Please check the pylon README > Troubleshooting:
https://www.baslerweb.com/fp-1584689425/media/downloads/software/pylon_software/README.txt 
This chapter is (most) relevant for you:
* Problem: My application can't see my camera. I'm not able to
  reconfigure the camera to make it visible again.

  Solution:
  First make sure you don't have a firewall enabled on the network
  adapter the camera is connected to.
  If you still can't see the camera, reverse-path filtering in the kernel may
  prevent your application from detecting the camera. This can happen if the 
  camera's IP address is not within the same subnet as the network adapter 
  the camera is attached to.
  To check whether filtering is turned on, run the following command:
    sysctl -a 2>/dev/null | grep '\.rp_filter'

  In the output look for the following lines:
   net.ipv4.conf.all.rp_filter=1
   net.ipv4.conf.eth1.rp_filter=1

m-binev, 2021-03-08:
@SaengthongStk ... just to avoid misunderstandings, here is the rest of the topic:
where "eth1" is the network adapter the camera is connected to.
  The "net.ipv4.conf.all.rp_filter" is a global switch that must be turned off.
  The "net.ipv4.conf.eth1.rp_filter" shows whether filtering for the specified
  network adapter is enabled. To disable filtering, you must first turn off
  filtering for "all" and the specific network adapter (in this
  sample "eth1"). Use the following commands to change the filtering behavior
  at runtime:
    sudo sysctl net.ipv4.conf.all.rp_filter=0
    sudo sysctl net.ipv4.conf.eth1.rp_filter=0

  Restart your application and check whether the camera(s) are
  detected. Reconfigure the camera(s) and use "Write Configuration" to make
  your changes persistent. You can re-enable filtering by executing the same
  commands but set a value of 1. If you want to turn off filtering permanently,
  you can edit the same values in /etc/sysctl.conf.

  From Linux kernel version 2.6.32 onwards, the rp_filter settings allow strict
  and loose filtering. To accept asymmetrically routed packets, modify /etc/sysctl.conf:
    net.ipv4.conf.default.rp_filter = 2
    net.ipv4.conf.all.rp_filter = 2

closed 2021-06-14 by pablo-quilez