basler/pylon-ros-camera #72: Grab and Save images not working

emanef13, 2021-03-12:
Hello,
I publish the below goal to the action server:
**~$ rostopic pub /pylon_camera_grab_and_save_as/grab_and_save_image_raw/goal camera_control_msgs/GrabAndSaveImageActionGoal "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs: 0
  id: ''
goal: {exposure_given: false, exposure_time: 3000.0, gain_given: false, gain_value: 0.0,
  gamma_given: false, gamma_value: 1.0, brightness_given: false, brightness_value: 0.0,
  exposure_auto: True, gain_auto: false, img_storage_path_and_name: '/home/manolis/image1.png'}" 
publishing and latching message. Press ctrl-C to terminate**

And I get the below error at the action server terminal:
**[ERROR] [1615557594.524704]: Exception in your execute callback: list index out of range
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/actionlib/simple_action_server.py", line 289, in executeLoop
    self.execute_callback(goal)
  File "/home/manolis/ws/src/pylon-ros-camera/pylon_camera/scripts/grab_and_save_image_action_server.py", line 80, in grab_and_save_img_raw_execute_cb
    self.grab_and_save_img_execute_cb(grab_and_save_img_goal,
  File "/home/manolis/ws/src/pylon-ros-camera/pylon_camera/scripts/grab_and_save_image_action_server.py", line 105, in grab_and_save_img_execute_cb
    cv_img = CvBridge().imgmsg_to_cv2(grab_imgs_result.images[0],
IndexError: list index out of range**

As I understand the grab_imgs_result.images is empty, but why? Am I right?
Any help would be very much appreciated. Thank you!

emanef13, 2021-03-15:
I think I need to first run the start_grabbing service. However, I get the following error in the pylon_camera_node termianal:
**[ERROR] [1615808346.473366169]: An exception while setting target image encoding to 'mono8' occurred: Node is not writable
[ERROR] [1615808346.473378920]: Error while start grabbing**

m-binev, 2021-03-15:
@emanef13 Can you check if your camera supports Mono8 pixel format at all? That is, some USB color cameras do not support monochrome pixel formats at all. In this case, you should try using Bayer8 instead.

emanef13, 2021-03-15:
Hi, thanks for your reply!
Yes the camera can support mono8 (Mono8 is not recognizable though, when I pass it as a parameter).
**[ INFO] [1615808915.912177845]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT']**

m-binev, 2021-03-15:
@emanef13 Okay, could you post the exact camera model name here, please?

emanef13, 2021-03-15:
Camera model: acA1920-155um

m-binev, 2021-03-15:
@emanef13 ok, thanks. Please make sure the camera is not running already. Actually, after StartGrabbing() is called, all critical payload parameters like Pixel Format, Width, Height etc. are turning from RW to RO.

emanef13, 2021-03-16:
@m-binev Hello, what do you mean by not running the camera? If I do not run the camrea_node the services are not available. Could you please specify that? Thanks for the help!

m-binev, 2021-03-16:
@emanef13 Hi, the forkflow for setting up a camera and grabbing images is as follows: 1) you first configure relevant parameters, e.g., width, height, pixel format, etc. 2) Then allocate ressources, e.g., buffer memory. 3) Then start image acquisition and grab images into the alocated buffers, which are reused for grabbing after releasing the memory. 
In pylon, 2) and 3) are pretty simplified and encapsulated in high level functions like StartGrabbing(). At point 2) though, all critical parameters like width, height, pixel format are locked and trurn from RW to RO. So, before setting e.g. pixel format, you have to make sure that StartGrabbing() has not been called, or that you call StopGrabbing().
On a side note: have you checked the following AppNotes: https://www.baslerweb.com/fp-1591878203/media/downloads/documents/application_notes/AW00149103000_Interfacing_Basler_Cameras_with_ROS~1.pdf

emanef13, 2021-03-18:
Thanks for your insights @m-binev! I was not able to use the start_grabbing service, however I made my own script that subscribes to the topic where raw images are published, and grabs an image by **img = rospy.wait_for_message(relative_topic, Image, 2)** .Then I am able to do my post-proccesing.

closed 2021-11-08 by pablo-quilez