basler/pylon-ros-camera #73: Install of driver

MartinSolk, 2021-03-17:
Hello, 
do you know if the ros driver works with pylon view 6.2 because if I try to install the pylon SDK API with 
'sudo rosdep install --from-paths . --ignore-src --rosdistro=$ROS_DISTRO -y' I become the failure that:
Could not find any pylon Installation with version 5 or greater
ERROR: the following rosdeps failed to install
  source: Failed to detect successful installation of [source: https://raw.githubusercontent.com/magazino/pylon_camera/indigo-devel/rosdep/pylon_sdk.rdmanifest]
I hope you can help me

m-binev, 2021-03-17:
@MartinSolk Hi, the current ROS driver expects pylon 5.2. If you want to use pylon 6.2, please check this new branch:
 https://github.com/basler/pylon-ros-camera/tree/pylon6-dev

MartinSolk, 2021-03-18:
Hi, I have new question. I am in the pylon 6 repository, but how to install this driver. If i work after the readme in this folder there is only the version for pylon5, then i tried to replace the camera_control_msgs and the pylon_camera in my /src/pylon-ros-camera but I still get the failure that no pylon5 version is found.

MartinSolk, 2021-03-18:
I hope you can help me or send me anything that the describes the installation :D

m-binev, 2021-03-18:
@pablo-quilez Can you help you with some guidelines? @MartinSolk this branch is right now only fine to use if you are using a Basler ace, ace2, dart USB or legacy ace GigE (not ace 2) camera. If this is not given on your side, please use the previous branch and install & use pylon 5.2 instead.

MartinSolk, 2021-03-18:
Currently I use this model: acA2440-75uc. I think it is a Basler ace. I am at the step to install the pylon SDK API with the command 'sudo rosdep install --from-paths . --ignore-src --rosdistro=$ROS_DISTRO -y'. There for i copied the updated folders, for pylon6  camera_control_msgs and pylon_camera in the path catkin_ws/src. Then i run the command. But i still get this mistake:
Could not find any pylon Installation with version 5 or greater
ERROR: the following rosdeps failed to install
source: Failed to detect successful installation of [source: https://raw.githubusercontent.com/magazino/pylon_camera/indigo-devel/rosdep/pylon_sdk.rdmanifest]
My question is why i get the mistake and how to avoid it.

m-binev, 2021-03-18:
@MartinSolk Hi, the new branch is in an experimental state yet, which is why, some things might not work as expected. I'm not sure why you need to use pylon 6.2 (5.2 should work just great for your camera model), but if you are in a hurry, I would suggest getting back to pylon 5.2 and use the "original" branch. Checking out this AppNotes might help either:
https://www.baslerweb.com/fp-1591878203/media/downloads/documents/application_notes/AW00149103000_Interfacing_Basler_Cameras_with_ROS~1.pdf

MartinSolk, 2021-03-18:
Hi,
thanks for the answer. I just thought that i need pylon 6.2 for more functions. but if 5.2 is good enough i will use this version.
But thank you, for the help

closed 2021-03-18 by MartinSolk