basler/pylon-ros-camera #74: Waiting For Trigger Signal/Error: Grab was not successful

twslankard, 2021-03-17:
Hi all,

I'm trying to implement hardware triggering of two GigE cameras so that they take a single frame whenever there is a pulse on line 3. To do this, I'm using the default Pylon ROS camera config, and then making a service call to set the trigger source to line 3 (int value 2).

Here's the method I implemented:
```
bool initializeCameras(ros::NodeHandle &n) {
    ros::ServiceClient trigger_source_client0 = n.serviceClient<camera_control_msgs::SetIntegerValue>("/cameranode0/set_trigger_source");
    ros::ServiceClient trigger_source_client1 = n.serviceClient<camera_control_msgs::SetIntegerValue>("/cameranode1/set_trigger_source");

    // set trigger source to line3
    camera_control_msgs::SetIntegerValue trigger_source_call;
    trigger_source_call.request.value = 2; // XXX this is from pylon_camera's pylon_camera_node.cpp, find a symbol
    if(!trigger_source_client0.call(trigger_source_call) || !trigger_source_client1.call(trigger_source_call)) {
        ROS_ERROR_STREAM("Couldn't set trigger source for cameras: " << trigger_source_call.response.message);
        return false;
    } else {
        ROS_INFO("Set trigger source for cameras successfully.");
    }
    return true;
}
```

It seems to work. I get a `image_raw` callback for each camera whenever line 3 gets a pulse. However, I'm seeing a lot of these errors in the logs:

```
[ERROR] [1616015132.304685759]: Waiting for Trigger signal
[ERROR] [1616015132.304724894]: Error: Grab was not successful
[ERROR] [1616015132.359765457]: Waiting for Trigger signal
[ERROR] [1616015132.359786272]: Error: Grab was not successful
```

I looked into the source, and found the following in `pylon_camera_base.hpp`:

```c++
template <typename CameraTrait>
bool PylonCameraImpl<CameraTrait>::grab(Pylon::CGrabResultPtr& grab_result)
{
    // XXX several lines omitted for brevity
    catch ( const GenICam::GenericException &e )
    {
        if ( cam_->IsCameraDeviceRemoved() )
        {
            ROS_ERROR("Lost connection to the camera . . .");
        }
        else
        {
            if ((! cam_->TriggerSource.GetValue() == TriggerSourceEnums::TriggerSource_Software) && (cam_->TriggerMode.GetValue() == TriggerModeEnums::TriggerMode_On))
            {
                ROS_ERROR_STREAM("Waiting for Trigger signal");
            }
            else
            {
            ROS_ERROR_STREAM("An image grabbing exception in pylon camera occurred: "
                    << e.GetDescription());
            }
        }
        return false;
    }
    // XXX several lines omitted for brevity
```

The caller of the method is what produces the "Error: Grab was not successful" message.

This leads me to my questions:

1. The code in `pylon_camera_base.hpp` handles non-software triggers as an exceptional condition. Does this mean non-software triggering is not fully supported?
2. What else do I need to configure on the camera, if anything? My assumption is that trigger mode should be "on", the trigger source should be line 3 (in my case), trigger selector should be "frame start".

m-binev, 2021-03-18:
@twslankard Hi, please check #28 and this branch:
https://github.com/Neel1302/pylon-ros-camera/tree/hardware-trigger

twslankard, 2021-03-18:
Thanks @m-binev I'll take a look.

FrancoisPicardDTI, 2022-02-07:
I am closing this issue as there's no activity here for one year, assuming that the latest updates have solved the problem.

closed 2022-02-07 by FrancoisPicardDTI