basler/pylon-ros-camera #75: Camera node dies as soon as camera matrix gets updated

emanef13, 2021-03-19:
**Hello,
Why my camera node dies as soon as I update the camera matrix K? I do update it by calling the set_camera_info service:**
$ rosservice call /pylon_camera_node/set_camera_info "camera_info:
  header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  height: 0
  width: 0
  distortion_model: ''
  D: [0]
  **K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]**
  R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  binning_x: 0
  binning_y: 0
  roi: {x_offset: 0, y_offset: 0, height: 0, width: 0, do_rectify: false}"

**When, I insert a value different that 0 in the K matrix, in the camera's node terminal says:**
[ INFO] [1615941352.735544232]: Camera is calibrated
[pylon_camera_node-1] process has died [pid 27803, exit code -11, cmd /home/manolis/ws/devel/lib/pylon_camera/pylon_camera_node __name:=pylon_camera_node __log:=/home/manolis/.ros/log/fbd05ee6-86ac-11eb-a608-071dbd7be759/pylon_camera_node-1.log].
log file: /home/manolis/.ros/log/fbd05ee6-86ac-11eb-a608-071dbd7be759/pylon_camera_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
But, whenever I insert a value different than 0 to any other argument of the 'set_camera_info' service, in the camera's node terminal says:
**[ INFO] [1615941237.782840476]: writing calibration data to /home/manolis/.ros/camera_info/camera.yaml**

and it is still running.

Any insight for the potential issue would be helpful. Thank you!

pablo-quilez, 2021-04-21:
Hello,

when you set the camera info, you need to provide the full correct description. This includes height, width, and the correct values for the different matrices. Here you can find an example of a calibration file: http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration. K, P and other matrices cannot be completely zeros: http://ksimek.github.io/2013/08/13/intrinsic/ 

You are probably right that the node should not die, but I think it is mostly related to mathematical incorrect input values.

Regards,
Pablo

emanef13, 2021-04-21:
Hi Pablo,

Thank you very much for your reply!
I found out that the camera node dies only the first time you assign a non zero value to the K matrix of the camera info .yaml file. As soon as the K matrix contains a non zero value, the parameters can be updated without the camera node dying every time you make an update. However, I have not figured out the reason why.

pablo-quilez, 2021-04-27:
Hi @emanef13 I keep it open. Thanks for the report. We will try to take a look to this.

FrancoisPicardDTI, 2022-10-31:
I am closing this issue as it seems that the problem is solved or at least is not occurring anymore.

closed 2022-10-31 by FrancoisPicardDTI