basler/pylon-ros-camera #76: A2440-75uc USB3 not present

MartinSolk, 2021-03-25:
Hi i try connect my camera. It can be opened in Pylon viewer but not in ros there the camera is not present. I opened ros in docker the workspace seems to be working. What could be answer. I hope you can help me.

m-binev, 2021-03-25:
@MartinSolk Hi, I suppose that docker container are not supported yet. Please try loading ROS without docker.

MartinSolk, 2021-03-25:
Ok, but a collegue of mine try it too it work at his station. But the steps were the same. any other possibility that it might can work, but there is only a small mistake that i didn't recognize

pablo-quilez, 2021-04-21:
Hi, is Pylon viewer in the host or in the guest system? Are you running the docker in priviledge mode? Are you sure docker has access to the /dev/usb directory of the host system? I think probably is related with docker priviledges / mounting. Regards.

closed 2021-04-27 by pablo-quilez