basler/pylon-ros-camera #78: input delay / high latency - basler acA2440-35uc

TKaluza, 2021-04-16:
On our system with Intel(R) Xeon(R) CPU E3-1505M v6 we have very heavy delays (several seconds, sometimes over 10 seconds) until the message arrives. We use a USB3 Basler acA2440-35uc camera for this. If we open PylonViewer directly on the same computer, it works fine. Via rqt_image_view (and no matter which topic we connect to) the message is delay very long. 
All settings are stored with in a User Settings, which is invoked while launching. The framerate is set to 20hz. No scaling is applied. The framerate of the received, but delayed message are okay.  We do not face this problem on a desktop pc setup.
On the first look I think this issues https://github.com/basler/pylon-ros-camera/issues/17 is fixed and does not relate to this problem anymore, or does ist? 
Are there any approaches to solving this problem?
Is there a option for software scaling within the pylon-ros-camera node?

FrancoisPicardDTI, 2022-10-31:
Hello @TKaluza 
Did you manage to solve your issue at the end? Maybe you can share here how you did solve it?

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI