basler/pylon-ros-camera #83: OpenCV version

e-team-t, 2021-06-01:
I am trying to compile this on a jetson nano with Jetpack 4.5,

make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/embross/catkin/devel/lib/libpylon_camera.so'.  Stop.

I have libopencv_objdetect.so.4.1.1 there instead, does pylon still uses opencv 3 ?

tuxx0rz, 2021-09-16:
I too had this issue on a Jetson Xavier AGX running the same jetpack version. I was able to get around many of these issues by installing the packages with apt-get until I got to it needing libopencv_aruco.so.3.2.0 which is part of the opencv3.2 contrib package which has a dependency of opencv3.2-core. I have not proceeded further as I am extensively using opencv4 and dont want to rip it out and replace with opencv3.2 without knowing the implications to my other opencv applications and if the downgrade is necessary at all. So yeah long story short I too would like to know if opencv3 is required and to let the devs know this appears to be affecting the whole Jetson lineup

carpenma, 2021-09-18:
While it isn't a perfect solution I was able to get around this problem by supplementing my existing opencv4 installation with the `libopencv3.2` package through apt (`libopencv-contrib3.2` was installed as well as a dependency).  That at least added all the required files so the ROS driver would build cleanly.

FrancoisPicardDTI, 2022-06-13:
Hello @e-team-t 
Pylon itself has never included OpenCV. This issue may be due to the fact you are using drag&bot, or the version of the pylon driver bound to it. Is that the case?

closed 2022-07-12 by FrancoisPicardDTI