basler/pylon-ros-camera #86: Two ACE2 cameras in master-slave with hardware synchronization trigger settings

BeSeeTek, 2021-08-05:
Hi Everyone I have two ACE2 Ethernet Cameras wired so that a Start Aquisition trigger is generated on Line 2 from the Master camera (Cam1).
The Salve Camera (Cam 2) is configured to use Line 2 as input and as aqusition trigger.
In pylon Viwer this also works when I change the framerate of the Master Camera the framerate of the Salve camera also changes.
## Master
[image]
[image]
## Slave
[image]
[image]

I have saved these camera configurations as user sets.
Loading the userset of the master camera works.
If the slave userset is loaded the following error occurs. And no frames are grabbed.
Is the procedure with the usersets correct?
How do I implement this set up reliably under ros?
Is it possible to deactivate the grabtimeout?

```
roslaunch pylon_camera pylon_cam2.launch 
... logging to /home/rosuser/.ros/log/4bec693e-f605-11eb-b847-151b1bc746ed/roslaunch-rosuser-Nuvo-7000-Series-18348.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rosuser-Nuvo-7000-Series:39181/

SUMMARY
========

PARAMETERS
 * /right/camera_frame: cam2
 * /right/camera_info_url: 
 * /right/device_user_id: cam2
 * /right/enable_current_params_publisher: False
 * /right/enable_status_publisher: True
 * /right/exposure: 10000.0
 * /right/frame_rate: 10.0
 * /right/gige/mtu_size: 1500
 * /right/image_encoding: mono16
 * /right/mtu_size: 3000
 * /right/shutter_mode: global
 * /right/startup_user_set: CurrentSetting
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    right (pylon_camera/pylon_camera_node)

auto-starting new master
process[master]: started with pid [18356]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4bec693e-f605-11eb-b847-151b1bc746ed
process[rosout-1]: started with pid [18366]
started core service [/rosout]
process[right-2]: started with pid [18373]
exposure is given and has value 10000
[ WARN] [1628178818.128175805]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1628178818.128567889]: Trying to open the following camera: cam2
[ INFO] [1628178818.426228375]: Found the desired camera with DeviceUserID cam2: a2A1920-51gmBAS
[ WARN] [1628178818.709586799]: No User Set Is selected, Camera current setting will be used
[ INFO] [1628178818.739263883]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT'] 
[ INFO] [1628178818.917091482]: CameraInfoURL needed for rectification! ROS-Param: '/right/camera_info_url' = '' is invalid!
[ WARN] [1628178818.917116830]: Will only provide distorted /image_raw images!
[ INFO] [1628178818.931393704]: Setting exposure to 10000, reached: 10000
[ INFO] [1628178818.947570402]: Startup settings: encoding = 'mono16', binning = [1, 1], exposure = 10000, gain = 0, gamma = 1, shutter mode = global
[ INFO] [1628178818.948620331]: Start image grabbing if node connects to topic with a frame_rate of: 10 Hz
[ INFO] [1628178818.948735808]: Camera not calibrated
[ INFO] [1628178853.768316960]: Cam supports the following [GenAPI|ROS] image encodings: ['Mono8'|'mono8'] ['Mono12'|'mono16'] ['Mono12p'|'NO_ROS_EQUIVALENT'] 
[ERROR] [1628178854.382157687]: An image grabbing exception in pylon camera occurred: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped.
[ERROR] [1628178854.382223676]: PylonCamera not ready because the result of the initial grab is invalid
[ERROR] [1628178854.382938680]: Tried to advertise a service that is already advertised in this node [/right/set_user_output_0]
[ERROR] [1628178854.384288790]: Tried to advertise a service that is already advertised in this node [/right/activate_autoflash_output_0]
[ERROR] [1628178854.385478402]: Tried to advertise a service that is already advertised in this node [/right/set_user_output_1]
[ERROR] [1628178854.386666499]: Tried to advertise a service that is already advertised in this node [/right/activate_autoflash_output_1]
[ERROR] [1628178854.387790430]: Tried to advertise a service that is already advertised in this node [/right/set_user_output_2]
[ERROR] [1628178854.389010879]: Tried to advertise a service that is already advertised in this node [/right/activate_autoflash_output_2]
```

farhad-dalirani, 2022-07-26:
Hi, 

I have two USB 3.0 Basler cameras,

I want to set one of them as master (generate trigger) and the other as a slave (receive trigger).

How should I do that?

Alex-Beh, 2023-04-27:
> I found a solution. left= master right=slave running this service calls in this order with the configuration saved as userSet 1 is working
> 
> ```
> rosservice call /right/stop_grabbing
> rosservice call /left/stop_grabbing
> rosservice call /left/select_user_set "value : 1"
> rosservice call /left/load_user_set
> rosservice call /left/start_grabbing
> rosservice call /right/select_user_set "value : 1"
> rosservice call /right/load_user_set
> rosservice call /right/start_grabbing
> ```

Now you can set the `startup_user_set` to `UserSet1` in the launch file [ROS1](https://github.com/basler/pylon-ros-camera/blob/08bb7cc9fc93849b4104a532ddf1e76e3ed51b5f/pylon_camera/launch/pylon_camera_node.launch#L7) or [ROS2](https://github.com/basler/pylon-ros-camera/blob/621f0d4580e784cd5c3fd77f152f1c98745b0f69/pylon_ros2_camera_wrapper/launch/pylon_ros2_camera.launch.py#L63) .