basler/pylon-ros-camera #99: 2 basler cameras synchronizes over USB

albeit07, 2021-12-07:
I have a research project, using 2 Basler cameras (both a2A1920-16ucPRO) to create a disparity map. I extract the images via ROS. For this I have subscribed to 2 Topics, where I regularly get images at 2 hertz. The problem is that these 2 images have a time difference of 5-10 milliseconds, but I need to get synchronously the 2 images, because iThe cameras are connected to my PC via USB. Is there any way to set something in ROS that I get 2 images in the same time? Or is there another solution? Here is my code:

import rospy
from sensor_msgs.msg import Image
from cv_bridge import CvBridge, CvBridgeError
import cv2
import time
import message_filters
Instantiate CvBridge

bridge = CvBridge()

def callback1(b1, b2):
    print("Aufruf beide Kameras!")
    try:
   
       cv2_img1 = bridge.imgmsg_to_cv2(b1, "bgr8")
       cv2_img2 = bridge.imgmsg_to_cv2(b2, "bgr8")

    except CvBridgeError, e:
           print(e)
    else:
  
       img_gray_left=cv2.cvtColor(cv2_img1, cv2.COLOR_BGR2GRAY) 
       img_gray_right=cv2.cvtColor(cv2_img2, cv2.COLOR_BGR2GRAY)
       dimension=(1024,440)
       resized1=cv2.resize(img_gray_left, dimension, interpolation=cv2.INTER_AREA)
       resized2=cv2.resize(img_gray_right, dimension, interpolation=cv2.INTER_AREA)
       cv2.imwrite('multilayer-stixel-world/build/myimages_ros/camera_image_left.pgm', resized1)
       cv2.imwrite('multilayer-stixel-world/build/myimages_ros/camera_image_right.pgm', resized2)
    

def main():
    rospy.init_node('image_listener', anonymous=True)
    image_topic1 = message_filters.Subscriber('/ace_RGB_Left/pylon_camera_node/image_raw', Image)
    image_topic2 = message_filters.Subscriber('/ace_RGB_Right/pylon_camera_node/image_raw',Image)
    ts = message_filters.ApproximateTimeSynchronizer([image_topic1, image_topic2],2, 0.005)
    ts.registerCallback(callback1)
    rospy.spin()

if _name_ == '_main_':
    main()

The message function "TimeSynchronizer" dont work, because the timestamps do not fit perfectly. So i use the function "ApproximateTimeSynchronizer" with a tolerance of 5 milliseconds. But that is not enough.

amarnathreddy0201, 2022-05-12:
Hi Sir/Mam,
I tried to synchronizes the Basler USB Cameras. there is no proper information about to synchronizing of two cameras.

I Studied lot of documents for the syn of two cameras. Every documentation saying using GigE cameras it's possible.

If you done synchronizing of 2 USB Basler cameras suggest me to do that.

Thanks,
ANR S

m-binev, 2022-05-12:
@albeit07 @amarnathreddy0201 Hi, the topic synchronisation of miltiple cameras is not so trivial, because 1) the different camera interfaces & protocols (GigE, USB) are different (and behave different), and 2) the use cases and requirements for synchronisation could vary. 
To 1) yes, the GigE protocol allows better and easier sync of multiple cameras(action commands, scheduled action commands, PTP (sync free run), however the USB protocol simply does not support similar features. Now, in all cases you can still Software Trigger to trigger the (serial) image acquisition of multiple cameras, while this would cause a delay ot 3-4 ms. You can also use Hardware Trigger to simultaneuosly trigger multiple cameras, or you can also connect the IO (output) lines of one camera to the (input) IO lines of other cameras. So, there are multiple options to do so.
To 2) when speaking of sync of cameras, from Machine Vision perspective (Factory automation, etc.) we typically consider the point when cameras are being triggerred, not when the images are received. If you expect to get images of multiple cameras syncronously - well, there are a lot of aspects to be considered... On ROS you have the general fact of images being additionally copied before being published. Depending of the image size, on system ressources, etc. this could cause some delays - even of a few seconds. 
I hope that sheds some more lights into the darkness))
Cheers,

farhad-dalirani, 2022-07-26:
How can I hardware trigger two Basler USB-3 cameras? I have the cable that connects both cameras to each other.

FrancoisPicardDTI, 2022-10-31:
Hello @albeit07 
I suppose you managed to solve your problem? Maybe you can share here how you did it?

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI