basler/pylon-ros-camera #103: Unable to select/change userset

elemecrobots, 2022-02-22:
Hi, I want to change the userset through rosservice call but I am unable to do so. I looked into the code and found out that there is no functionality implemented to change the userset through rosservice call. can anyone please confirm this? Ideally, I would like to change the userset through rosservice call /pylon_camera_node/select_user_set "value: 3" for example.

nickzherdev, 2022-03-10:
Hey. I didn't try it using service, but regular launching works well:

```
<include file="$(find pylon_camera)/launch/pylon_camera_node.launch">
            <arg name="node_name" value="left" />
            <arg name="config_file" value="$(find pylon_camera)/config/front_left_settings.yaml" />
            <arg name="startup_user_set" value="UserSet1" /> 
    </include>
```
Before that in pylonviewer go to usersets field and press "load" to chosen userset the setup you want to use.

FrancoisPicardDTI, 2022-03-23:
Hello @elemecrobots 
Did @nickzherdev's solution solve your problem?

closed 2022-04-19 by FrancoisPicardDTI