basler/pylon-ros-camera #104: PylonCamera not ready because the result of the initial grab is invalid

nickzherdev, 2022-03-10:
Hello!
I have a stereo camera setup with "dragchain" synchronization, where one camera is set up as master with software trigger, and the other camera is connected by Line2 and uses hardware triggers from master camera.

Here is the launch file:
```
<launch>

    <include file="$(find pylon_camera)/launch/pylon_camera_node.launch">
            <arg name="node_name" value="right" />
            <arg name="config_file" value="$(find pylon_camera)/config/front_right_settings.yaml" />
            <arg name="startup_user_set" value="UserSet1" /> 
    </include>

    <include file="$(find pylon_camera)/launch/pylon_camera_node.launch">
            <arg name="node_name" value="left" />
            <arg name="config_file" value="$(find pylon_camera)/config/front_left_settings.yaml" />
            <arg name="startup_user_set" value="UserSet1" /> 
    </include>

</launch>
```
Unfourtunately, sometimes a get an error when launching the salve camera (which receivers hardware triggers from the master camera):

> [ INFO] [1646155461.688474027]: Camera is calibrated
> [ERROR] [1646155462.529302276]: An image grabbing exception in pylon camera occurred: Grab timed out.
> [ERROR] [1646155462.529419687]: PylonCamera not ready because the result of the initial grab is invalid

And sometimes both cameras launch flawlessly. Statistics is around 50-50
Here are the ros logs:
[good_basler.md](https://github.com/basler/pylon-ros-camera/files/8223846/good_basler.md)
[bad_basler.md](https://github.com/basler/pylon-ros-camera/files/8223847/bad_basler.md)

Could you suggest what could be the problem?
Do I miss some delay?

m-binev, 2022-03-10:
@nickzherdev Hi, may you describe your setup, please? That is, what camera models? any Switch? etc. I suppose, you experience IP packet collisions, packet losses and image losses and would need to optimize camera parameters like PacketSize, Inter-Packet Delay, Frame Transmission Delay. Cheers

nickzherdev, 2022-03-15:
@m-binev I'm sorry, of course. 
Ace2 1920-51 gc BAS, no switch.
gige:
  inter_pkg_delay: 5607
gige:
  mtu_size: 8192

Frame Transmission Delay is 0 for both cameras, they need to be synchronized.

nickzherdev, 2022-04-12:
Hello!  May I ask you to suggest which parameters should I tune further?

m-binev, 2022-04-12:
@nickzherdev Hi Nick, beg a pardon for the delay. To be honest, as your setup is a little advanced and the exact configuration and root cause are still unclear (to me), it is difficult to say, which parameter(s) should be exactly touched/set in order to get a 100% working setup. I'd suggest that you first test your setup in/with pylon Viewer (without ROS) in order to make sure, the setup works itself. However, as the setup appears to work in about 50% of the cases, it cannot be a general issue, I think. 
In my opinion, it is either 1) something related to overtriggering or 2) image losses/timeouts. 
To 1) using auto exposure while triggering a camera with constant frame rate might be tricky - maybe setting some fixed exposure time (e.g., 5000 us) that cannot influence the triggering could be tried out for test purposes. Also, as I am not sure, which trigger modes, etc. are used, you could consult the Basler online documentation (https://docs.baslerweb.com/a2a1920-51gcbas) and re-check your camera trigger settings. 
To 2) I would suggest to check and set the s.c. Frame Transmission Delay (FTD) for both cameras to some value. As I don't have the same camera model as yours at hand, I cannot tell you exact values. You could do the following in pylon Viewer though: for cam1 increase the FTD to a value that starts decreasing the resulting frame rate (in your case it appears to be 15 Hz). For cam2: take the FTD value of cam1 + 8192 (mtu) + 18  (all in ticks). In all cases, check the Statistics parameters - especially Failed Buffer Count, Buffer Underrun. 
I hope that helps. Cheers

farhad-dalirani, 2022-07-28:
Can you please give your ros config and ros long for both cameras? I have a similar problem which I can not solve: https://github.com/basler/pylon-ros-camera/issues/125

nickzherdev, 2022-11-29:
@FrancoisPicardDTI Hello!  No, problem was not solved, we moved to clean hardware sync

closed 2024-02-22 by FrancoisPicardDTI