basler/pylon-ros-camera #107: Using external trigger could only get a topic up to 20hz

TheConquerK, 2022-04-11:
I'm using basler ac1920 uc155 camara to work for a hardware syncronization.
It's well performed on pylon-viewer which could be triggered up to 50hz.
While using pylon-ros-camara ,selecting grab-stratagy 1 ,could only trigger a topic up to 20hz

m-binev, 2022-04-11:
@TheConquerK Hi, comparing pylon Viewer (optimize for maximum performance) with the "pylon-ros-camara" driver is not quite fair, especially while keeping in minde that additional copy operations occur in ROS. That is, depending on the maximum possible camera frame rate, camera resolution, used pixel format, etc., you may have cases, where you would not be possible to get the maximum of the camera on ROS (also due to extra copying data). Now,  grab-stratagy =1 (latest image only) is usually a good strategy to use when the image post-processing (e.g., display, save, run calculations on the image) takes longer than the image acquistion. In this cases, you would like to work with the latest image only and all others will be skipped by the pylon InstantCamera class. Thus, this stratega is not the best one to get the maximum frame rate and all images. What you can try out  is, Pixel Format= Bayer8 vs RGB8; Grab Strategy= 0 (one by one, i.e., to get all images). However, as said before, under circumstances you might not get the same performance as in pyon Viewer eventually. Cheers.

TheConquerK, 2022-04-11:
@m-binev Thanks for your reply . I think that I know how to get a higher rate now, I will try on that.
And there's one more question, I tried that Grab Strategy= 0 will get a higher rate topic on ROS . Every things are ok when I use camera internal clock. But if I use external clock (using MCU to create  square wave)  ,it will cause a big time delay(  almost 2 seconds ).

m-binev, 2022-04-11:
@TheConquerK HI, I'm glad to help out a bit)) Well, it sounds as if you might be overtriggering the camera or something... (checking the Line Debouncer & the trigger modes might be a good idea). As this is difficult to debug virtually, I can recommend you reading the camera dosumentation: https://docs.baslerweb.com/trigger-selector 
Regarding triggering itself, this could be a reason for you to contact your Basler (or responsible distributor) support contact person. Cheers

TheConquerK, 2022-04-12:
@m-binev I will check my camera settings. Thanks for your help , I will soon close this issue.

closed 2022-04-12 by TheConquerK

farhad-dalirani, 2022-07-27:
I have a similar question, do you have your launch and config file for ROS?

https://github.com/basler/pylon-ros-camera/issues/125