basler/pylon-ros-camera #108: Pylon_camera does not build on jetson AGX

rouceto1, 2022-04-13:
Hi i have issues building the pylon_camera for 
Jetson AGX (Jetpack 5.0)
Ubuntu 20
ROS Noetic

The original error 
 `libpylon_camera.so: undefined reference to 'cv::Mat::Mat()' `
 
 I have reinstalled two different versions of OpenCV (4.5.2 and 4.5.4) the latter one with and without CUDA. 
 None of this helped.
 
 When i run the rosdep install from the repo it returns
 `pylon_camera: Cannot locate rosdep definition for [pylon]` but i assume it is not the issue here. 
 
 Pylon was installed using the deb package with version 6.2.0.

FrancoisPicardDTI, 2022-07-28:
Hello @rouceto1 
Pylon itself has never included OpenCV. This issue may be due to the fact you are using drag&bot, or the version of the pylon driver bound to it.
The "libpylon_camera.so" library is not a native pylon library, thus, it must be coming from drag&bot. Maybe you can try the ROS2 version of the pylon driver?
I am closing the issue.

closed 2022-07-28 by FrancoisPicardDTI

peci1, 2022-09-21:
We've finally solved it with @rouceto1 . This is a problem caused by NVidia's bad ideas about package naming. Jetpack 5.0 installs package `libopencv-dev` in version 4.5, which overrides the Ubuntu-ditributed `libopencv-dev` 4.2. But it does not override the other Ubuntu-distributed packages, so you end up with a system that has half of its opencv 4.2 and the other half 4.5.

To resolve the issue, we did the following:

1. `dpkg -r --force-depends opencv-licenses` # not needed and conflicts with Ubuntu opencv
2. Create file `/etc/apt/preferences.d/fix-opencv` with the following content:

```
Package: libopencv-dev
Pin: release o=Nvidia,a=stable,n=r35.1,l=L4T Jetson r35.1,c=main,b=arm64
Pin-Priority: 9
```

3. `sudo apt remove libopencv-dev` # uninstall the evil 4.5 version; it will want to uninstall some packages; copy their list and allow that
4. `sudo apt install libopencv-dev` # this installs the correct 4.2 version; also install the other packages from previous step
5. clean and rebuild workspaces

rouceto1, 2024-02-27:
The problem is still persistent in newer jetpacks. 
To get proper content of the file run 
`apt-cache policy`
and copy the proper n and l parameter from the L4T-jetson line