basler/pylon-ros-camera #109: error: ‘pylon_camera::PylonCameraImpl<pylon_camera::USBCameraTrait>::CBaslerInstantCameraT {aka class Pylon::CBaslerUniversalInstantCamera}’ has no member named ‘BslLightSourcePreset’; did you mean ‘LightSourcePreset’?

knightdby, 2022-05-20:
Camera: acA1920-40gc
Platform: jetson xavier, ARM64
Pylon: 5.2.0

`src/pylon_camera/include/pylon_camera/internal/impl/pylon_camera_base.hpp:2503:83: error: ‘Basler_UniversalCameraParams::BslLightSourcePresetEnums’ has not been declared
                 cam_->BslLightSourcePreset.SetValue(Basler_UniversalCameraParams::BslLightSourcePresetEnums::BslLightSourcePreset_Off);`

m-binev, 2022-05-20:
@knightdby Hi, I recommend using the master branch, which includes pylon 6.2. Cheers

knightdby, 2022-05-20:
when i use pylon 6.2, then error:
[image]

m-binev, 2022-05-20:
@knightdby can you check the latest Readme: https://github.com/basler/pylon-ros-camera/tree/galactic 
Especially:
GigE Devices
No connection with connected camera
To be sure to be able to connect to a specific camera, its network configuration must be manually set through Basler's pylon IP configurator. To do so, click on the camera in the list of connected devices, select the Static IP option, set an IP Address within the same range as the one of your computer, and set the same Subnet Mask as the one from your computer.

and:
Setting device user id

FrancoisPicardDTI, 2022-05-20:
@knightdby can you post the whole from the moment you launch the node please?

knightdby, 2022-05-20:
@m-binev thanks for your reply,I haven't set the use ID before. Wait for me to try,thanks.

knightdby, 2022-05-20:
@FrancoisPicardDTI Thank you for your reply,The whole terminal log is as follows：
[image]

FrancoisPicardDTI, 2022-05-23:
Hello @knightdby 
Thank you for the logs,
As it is specified in the logs, the camera model you're using, alA1920-40gc, is a limited low-cost camera and is not compatible with the current implementation of the ROS driver. Due to missing camera features, the current implementation does not recognize the camera model, preventing thus using this camera. We recommend using regular Basler cameras instead.

knightdby, 2022-06-06:
@m-binev 
Thank you for your suggestion. I now use ROS image transport compression for data transmission, which can solve the problem of insufficient bandwidth, but it can not be completely solved.
Then, I would like to ask you that the exposure time cannot be set in the ROS driver:
[image]
Camera: acA1920-40gc
Platform: jetson xavier, ARM64
Pylon: 5.2.0 and 5.1.0
Looking forward to your reply,Best wishes to you!

m-binev, 2022-06-08:
@knightdby Hi, I'm glad I could help. As per the exposure time: why don't you use the latest implementation based on pylon 6.2/6.3? This should make lots of things easier. Otherwise, I don't see what's the exact parameter call you make, but this camera should support the parameter ExposureTimeAbs, which controls the exposure time. Of course, this one is not writable while ExposureAuto is set, i.e., make sure you are not running the auto exposure. Cheers

knightdby, 2022-06-09:
@m-binev 
Thank you for your reply. More information is as follows:
[image]
[image]
[image]
Neither exposure time nor brightness can be written to the camera.

m-binev, 2022-06-09:
@knightdby Hi, I don't have hardware to test that, but looking at the raw code, a call to setting the brightness (this is no camera parameter), will try to set AutoExposure and AutoGain. That is, after that you won't be able to write to exposure time or gain. Also, I still recommend you using the latest driver version instead)). To troubleshoot things, please power off the camera and try to disable the brightness and any automatic settings. Cheers

knightdby, 2022-06-20:
@m-binev Thanks for your help, everything is all right

closed 2022-06-20 by knightdby