basler/pylon-ros-camera #110: On Ace2 basler 1920-51gc upper exposure value is not limited to chosen value

nickzherdev, 2022-05-24:
**Problem description**
When exposure_auto is set to true, brighntess of the image is ~254, accroding to CurrentParams and the image itself, though upper limit is set and target brightness is twice as low. The resulted ExposureTime is 30'000 (or 100'000 if I load default UserSet in pylon). 
I have the similar problem with acA 1300-60 gc, it also doesn't hold the upper limit, but with the resulted brightness is not that high.

**Installed versions**
```
pylon 6.2.0.21487-deb0
pylon-ros-camera commit hash 56e8d25b291325eca196dc6ed31bfb7e5e79bc1f

```
**default.yaml configs**
```
image_encoding: "rgb8"
exposure: 500.0
brightness: 100
brightness_continuous: true
exposure_auto: true
gain_auto: false  (also tried "true", result is mostly the same)
auto_exposure_upper_limit: 1000.0

```
Other values are default.

What can be the problem and how to fix it?  For now I turned off auto modes and set some manual values, but they are responsive to lightning changes, obviously.

FrancoisPicardDTI, 2022-11-11:
Hello @nickzherdev 
I checked the code and I can tell you that this exposure upper limit is only set at the camera instance level when the startup user set parameter is set to "Default". Can you try that and see if it solves your problem?
If you are starting the driver through its launch file, change this parameter in it, as the launch file has the priority over the parameter server.

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI