basler/pylon-ros-camera #111: Any support for CSI cameras?

mbrymer, 2022-05-25:
Hi, I have the daA2500-60mci camera from the Jetson developer kit and am looking to integrate it into a ROS environment. I had been under the impression that my configuration would be supported with the included drivers but after getting the node running I'm getting the following error:

The detected camera type is: BaslerGTC/Basler/daA2500_60mci. Only 'BaslerUsb' and 'BaslerGigE' types are supported by this driver for now!

Is there any part of this node that I could somehow salvage or am I SOL and forced to either write my own node from scratch or find another camera?

m-binev, 2022-05-25:
@mbrymer Hi, the Readme clearly says that only GigE and USB 2D cameras are supported yet. Sorry(( and no, supporting CSI cams would require additional development for the given hardware, which might be quite costy... I could recommend using the same dart model with USB interface instead - or ace/ace 2, which are much moch powerful. Cheers

mbrymer, 2022-05-25:
Ack, okay that's what I thought. No worries, thanks for your quick response!

closed 2022-05-25 by mbrymer