basler/pylon-ros-camera #113: Problem with catkin_make

antun008, 2022-06-05:
Hello,
when I execute catkin_make,  I get this error:
[image]

Can someone help me plssssss

nickzherdev, 2022-06-06:
It looks like you don't have OpenCV installed
`sudo apt-get install libopencv-dev`

FrancoisPicardDTI, 2022-06-07:
@antun008 hello,
It looks like you're are using OpenCV somewhere. The pylon driver does not use it originally so I think your error is related to your own application. Try indeed to install OpenCV to see if it solves your problem.

FrancoisPicardDTI, 2022-06-13:
Hello @antun008  
So, just for confirmation, pylon itself has never included OpenCV. This issue may be due to the fact you are using drag&bot, or the version of the pylon driver bound to it. Is that the case?
The "libpylon_camera.so" library is not a native pylon library, thus, it must be coming from drag&bot.

closed 2022-06-13 by antun008