basler/pylon-ros-camera #116: Low FPS in ROS a2A1920-51gcPRO

TRA55A, 2022-06-19:
Hi,

When I use the Pylon viewer I get 30FPS but when using the ROS package I get 17FPS, I tried to change some settings on the yaml file but with no success.

.pfs output from pylon viewer:
[a2A1920-51gcPRO_40135680.pfs.txt](https://github.com/basler/pylon-ros-camera/files/8935397/a2A1920-51gcPRO_40135680.pfs.txt)

config file from ROS package:
[default.yaml.txt](https://github.com/basler/pylon-ros-camera/files/8935404/default.yaml.txt)

Any ideas on what I need to change on the ROS package in order to get the same performance?

Thanks

m-binev, 2022-06-20:
@TRA55A Hi, there are some other open/closed issues related to frame rate here. Maybe you want to check them? In a nutshell, you cannot expect getting the same frame rate on ROS as in pylon Viewer, because you have additional image copy operations on ROS. Also, the maximum (or higher) frame rates are only possible, when running a camera in free-run or hardware trigger mode. Using a software trigger (default mode on ROS) would result in about the half of the possible frame rate. Cheers

TRA55A, 2022-06-21:
Thanks, I read most of the posts, I noticed some users noted issues with the enable_current_params_publisher I will try to change it to false.

How do I change the trigger mode? 
In my YAML file, I changed the grab_strategy to 2.
Should I change another parameter?

TRA55A, 2022-06-22:
Update: after changing  enable_current_params_publisher  to false, I'm getting 25-28 FPS.

Thanks!

closed 2022-06-22 by TRA55A