basler/pylon-ros-camera #120: Fix huge latency when free running or using AutoFunction (solution Inside)

mistermult, 2022-07-05:
# Problem
1. Configure with Pylon Viewer the camera and save in User Set 1:
- set AutoFunction (automatic Exposue and automatic Gain) 
- set frame rate, e.g. to 10 Hz
2. Choose User Set 1 in Pylon ROS Node
- set frame rate to 10Hz in config
Note that frame rate cannot be set higher than 10Hz because when the node starts it will ask the camera for max possible framerate and only allow this at maximum.
3. Huge delay, e.g. 10s, is observed

# Reason for Problem
Camera seems to trigger, e.g. with 10.2Hz but images are only grabbed with 10Hz, because the frame rate is the rate for the main loop which grabs the images. So a queue will get build up which is the reason for the delay.

# Debugging and Fix
- modified code with debug output to confirm that software trigger function `cam_->ExecuteSoftwareTrigger()` (https://github.com/basler/pylon-ros-camera/blob/caff7fe26095dfb0bb803873b2fbaf32f1daf64b/pylon_camera/include/pylon_camera/internal/impl/pylon_camera_base.hpp#L490) is really NOT called
- confirmed that maximal rate for main loop can maximally set to 10Hz (in example above) https://github.com/basler/pylon-ros-camera/blob/caff7fe26095dfb0bb803873b2fbaf32f1daf64b/pylon_camera/src/pylon_camera/main.cpp#L48 even if higher frame_rate is set in config/ros parameters:

- if I hardcode the loop rate in the main loop to 10Hz (if framerate in camera user set is 10Hz) I get huge latency (10s), i.e. if I replace https://github.com/basler/pylon-ros-camera/blob/caff7fe26095dfb0bb803873b2fbaf32f1daf64b/pylon_camera/src/pylon_camera/main.cpp#L48 by
```c++
ros::Rate r(10.0);
```
- if I hardcode the framerate in the main loop to e.g. 100Hz (if framerate in camera user set is 10Hz) I get no latency, i.e. if I replace https://github.com/basler/pylon-ros-camera/blob/caff7fe26095dfb0bb803873b2fbaf32f1daf64b/pylon_camera/src/pylon_camera/main.cpp#L48 by
```c++
ros::Rate r(100.0);
```
This is the fix: Setting the grab rate higher than the maximal possible frame rate of the free running camera.

# Proposed Solution
1. Either do not truncate the frame rate (which configures the `ros::Rate`) given by ROS parameters by maximal frame rate reported from camera,
2. Or add an additional parameter (e.g. `grab_rate`) to configure the `ros::Rate` of the main loop independently. If the parameter is not set one could default initialize it with the normal `frame_rate` parameter.

Please let me know what you think about this idea and if that can be modified/merged in the official branch. I do not mind if you do the modifications yourself.

seajayshore, 2022-10-05:
Thank you so much for this excellent issue + fix proposal! It's helped solve a tonne of down-stream issues with my system!

I actually used the following in main.cpp in the end:
`ros::Rate r(pylon_camera_node.frameRate() * 3);`

So the spin rate is still somewhat based on frame rate and not unnecessarily fast.

FrancoisPicardDTI, 2022-10-31:
Hello @mistermult 
Yes please, suggest a pull request and I'll have a look at it. Thank you!

closed 2024-02-22 by FrancoisPicardDTI