basler/pylon-ros-camera #121: Set binning not working via ROS with acA4600-7gc

sebastian-ruiz, 2022-07-25:
My camera is acA4600-7gc. According to the [basler website](https://docs.baslerweb.com/binning), the camera supports binning.

I can set the "decimation horizontal" and "decimation vertical" to 2 in the pylon viewer under AOI controls. I can save this to a userset and this all works.

I assume decimation is the same as binning.

**Problem**: I would like to be able to set the binning programmatically via ROS. 

When I try and set the binning mode to 2 using a rosservice call I get:
```
$ rosservice call /pylon_camera_node/set_binning "target_binning_x: 2
target_binning_y: 2"
reached_binning_x: 1
reached_binning_y: 1
success: True
```
It doesn't set it to 2, only to one.

I also try and set binning in the config/default.yaml file, like so:
```yaml
binning_x: 2
binning_y: 2
```
I get the following warning in ROS:
```
[ WARN] [1658762120.127700397]: Camera does not support binning. Will keep the current settings

[ WARN] [1658762120.128583094]: The image width of the camera_info-msg will be adapted, so that the binning_x value in this msg remains 1

[ WARN] [1658762120.128616336]: Camera does not support binning. Will keep the current settings

[ WARN] [1658762120.129438373]: The image height of the camera_info-msg will be adapted, so that the binning_y value in this msg remains 1
```
This also doesn't work.

It would be great to know what I am doing wrong, thanks!

FrancoisPicardDTI, 2022-11-10:
Hello @sebastian-ruiz 
This model does indeed support binning but the binning is done by the sensor itself.
This is far from being the case each time. It is normally done on the camera FPGA.
You need to set one more parameter that is probably not available on ROS yet:
`camera.BinningSelector.SetValue(BinningSelector_Sensor);`

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI