basler/pylon-ros-camera #125: Hardware Synchronisation of Two USB 3.0 Basler Cameras

farhad-dalirani, 2022-07-27:
Hello,

# problem

I use two USB 3.0 Basler cameras as a stereo system on a car. Since the car moves, I need to synchronize two cameras. For this reason, I bought a 6-pin cable that connects cameras to each other. 

I want the left camera to be the master and generate a trigger, and I want the right camera to be a slave and start to capture by receiving a trigger.

I did the setting in pylon viewer, which works in pylon viewer. I stored the setting for two cameras in their user set 2. When I want to work with ROS, I load the user set 2 for each camera. However, I get an error.

To give you more detail, in the following, I provided the information:
- the setting for the left camera in pylon viewer, which I save in user set 2
- the setting for the right camera in pylon viewer, which I save in user set 2
- ROS launch file
- ROS config for the left camera
- ROS config for the right camera
- Error message and information related to image topics

## the setting for the left camera in pylon viewer, which I save in user set 2
[image]
[image]
[image]
[image]
[image]
[image]

## the setting for the right camera in pylon viewer, which I save in user set 2
[image]
[image]
[image]
[image]
[image]
[image]
[image]

## ROS launch file
[image]

## ROS config for the left camera
[image]

## ROS config for the right camera
[image]

## Error message and information related to image topics
[image]
[image]

Thank you for your time,

farhad-dalirani, 2022-07-27:
If you need more detail let me know.

FrancoisPicardDTI, 2022-10-31:
Hello @farhad-dalirani 
Difficult to see what is wrong here. For the sake of testing here, can you comment in your ROS config files the parameters related to the brightness (brightness and brightness_continuous)? What is happening when you try then to start your process?

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI

rows-of-houses, 2023-05-05:
It seems like I managed to synchronize two Basler USB 3.0 cameras with a hardware trigger in ROS Noetic. I have two daA1280-54uc cameras.
I didn't change configuration of the cameras in Pylon viewer. I don't use any user presets. I provide all parameters either as ROS parameters or calling ROS services in launch file, like in mine (I can't paste it all for unknown reasons, so I provide screenshot)

`<?xml version="1.0"?>
<launch>
	<group ns="camera">
		<arg name="respawn" default="false" />
		<arg name="debug" default="false" />
		<arg name="mtu_size" default="1500" />
		<arg name="startup_user_set" default="CurrentSetting" />
		<arg name="enable_status_publisher" default="true" />
		<arg name="enable_current_params_publisher" default="true" />
		<arg name="config_file" default="$(find pylon_camera)/config/default.yaml" />
		<arg name="frame_rate" default="20"/>
		<arg name="binning_x" default="2"/>
		<arg name="binning_y" default="2"/>
		<arg unless="$(arg debug)" name="launch_prefix" value="" />
		<arg     if="$(arg debug)" name="launch_prefix" value="gdb -ex run --args" />
		
		<node name="left" pkg="pylon_camera" type="pylon_camera_node" output="screen"
		      respawn="$(arg respawn)" launch-prefix="$(arg launch_prefix)">
		    
		    <param name="gige/mtu_size" value="$(arg mtu_size)"/>
		    <param name="device_user_id" value="left"/>
		    
		    <param name="enable_status_publisher" value="$(arg enable_status_publisher)"/>
		    <param name="enable_current_params_publisher" value="$(arg enable_current_params_publisher)"/>
		    <param name="frame_rate" value="$(arg frame_rate)"/>
		    <param name="binning_x" value="$(arg binning_x)"/>
		    <param name="binning_y" value="$(arg binning_y)"/>
		    
		</node>

		<node name="right" pkg="pylon_camera" type="pylon_camera_node" output="screen"
		      respawn="$(arg respawn)" launch-prefix="$(arg launch_prefix)">
		    
		    <param name="device_user_id" value="right"/>
		    <param name="gige/mtu_size" value="$(arg mtu_size)"/>
		    
		    <param name="enable_status_publisher" value="$(arg enable_status_publisher)"/>
		    <param name="enable_current_params_publisher" value="$(arg enable_current_params_publisher)"/>
		    <param name="frame_rate" value="$(arg frame_rate)"/>
		    <param name="binning_x" value="$(arg binning_x)"/>
		    <param name="binning_y" value="$(arg binning_y)"/>
                    
		</node>
		<node pkg="rosservice" type="rosservice" name="reverse_x" args="call /camera/right/set_reverse_x true"/>
		<node pkg="rosservice" type="rosservice" name="reverse_y" args="call /camera/right/set_reverse_y true"/>
		<node pkg="rosservice" type="rosservice" name="set_trigger_source" args="call /camera/right/set_trigger_source 1"/>
		<node pkg="rosservice" type="rosservice" name="set_line_selector_r" args="call /camera/right/set_line_selector 0"/>
		<node pkg="rosservice" type="rosservice" name="set_line_selector_l" args="call /camera/left/set_line_selector 0"/>
		<node pkg="rosservice" type="rosservice" name="set_line_mode_r" args="call /camera/right/set_line_mode 0"/>
		<node pkg="rosservice" type="rosservice" name="set_line_mode_l" args="call /camera/left/set_line_mode 1"/>
		<node pkg="rosservice" type="rosservice" name="set_line_source" args="call /camera/left/set_line_source 0"/>
		<node pkg="rosservice" type="rosservice" name="set_grabbing_strategy_l" args="call /camera/left/set_grabbing_strategy 1"/>
		<node pkg="rosservice" type="rosservice" name="set_grabbing_strategy_r" args="call /camera/right/set_grabbing_strategy 1"/>
				
		
	</group>
	
</launch>`
[image]

I checked two cameras with a timer on 60 Hz phone screen, it seems like they are synchronized with a accuracy of 0.016s.
[image]

I am setting up a stereo-inertial system for a robot navigation. I use Kalibr ROS package for calibration of the system. According to Kalibr, time offset between two frames is about 0.00015 s. That means that frames are really synchronized.

jfaure6, 2025-01-28:
Hello @rows-of-houses ,

I am facing issues when I want to synchronize 2 acA1440-220uc basler cameras.
My setup is a few different compared to yours, as I am triggering my two cameras from the same signal coming from an arduino. The signal triggers on rising edge for both cameras, and I am using the ROS1 driver on ROS Noetic like you. I just set up the input port of the Arduino on Pylon viewer for both cameras.
The issue I am facing is that it is not synchronyzed at high frequencies, and I can see it by putting a timer as you did, and seeing different value on the timer from a same trigger. Also, I am wondering how did you use Kalibr ROS package to see the time offset.

If you have any clue...