basler/pylon-ros-camera #127: Issue: PylonTerminate not called upon Ctrl+C or rosnode kill

ivan140, 2022-08-03:
Hi
I'm using Basler ac USB cameras and observe the following issue:
After launching the `pylon_camera` node and killing it (Ctrl+C or rosnode kill), pylonviewer is not able to grab frames from the camera.
If I execute `set_device_user_id` script, which does execute PylonTerminate, pylonviewer is again capable of grabbing frames from the camera.
Maybe a custom [SIGINT handler](http://wiki.ros.org/roscpp/Overview/Initialization%20and%20Shutdown#Custom_SIGINT_Handler) would make sense here.

FrancoisPicardDTI, 2022-10-31:
Hi @ivan140 
Isn't it due to the Trigger Mode parameter in the Acquisition Control panel? The ROS driver is setting it to "On". If you set it back to Off in the pylon viewer, does it begin grabbing in free run mode?

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI