basler/pylon-ros-camera #129: basler ip camera config not work

xuchengithub, 2022-09-26:
i use basler acA2440-20gc and usb camera , i setup branch ros2 galactic in docker.when i run ros2 cmd camera is working but some config is not working well.
1.when i set basler acA2440-20gc image_encoding: "rgb8" but get setup encoding = 'mono8'(usb camera is working well)

[pylon_ros2_camera_wrapper-1] 1664175740.910657094 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Startup settings: encoding = 'mono8', binning = [1, 1], exposure = 43000, gain = 0.45, gamma = 1, shutter mode = global

2both camera (ip and usb camera )set
 binning_x: 2
 binning_y: 2
is not work

can you help me ,thanks

xuchengithub, 2022-09-27:
> @xuchengithub Hi, color cameras usually do not support binning. Please use the pylon Viewer to check our the supported parameters of your camera first, or refer to the Basler product documentation: https://docs.baslerweb.com/ Cheers

thanks for you response.
you means to see ,if the camera can use basler ros2 .i use basler view sample code c++ get the images well.

i run pylonview in docker ,and see the image of acA2440-20gc  ,images is color .
but when i use basler ros2 code and just use the sample code . set
camera_frame: pylon_camera
device_user_id
image_encoding: "rgb8",
not change other things .
i only get mono8 images
im tiro of ros2，so i can not konw how to change the code。

xuchengithub, 2022-09-27:
> thanks for you response. you means to see ,if the camera can use basler ros2 .i use basler view sample code c++ get the images well.
>
> i run pylonview in docker ,and see the image of acA2440-20gc [...]

ok i find i just dont try
bayer_rggb8
is working for me

FrancoisPicardDTI, 2022-09-27:
I am closing the issue then.

closed 2022-09-27 by FrancoisPicardDTI