basler/pylon-ros-camera #131: Off by one error calling GrabImages.action

rshanor, 2022-10-16:
I am not publishing images, just using the GrabImages.action interface. I have noticed an off by one error where the frame is actually the previous one I would have expected to capture. Anyone else run into this issue?

FrancoisPicardDTI, 2022-10-31:
Hello @rshanor 
I am not noticing anything with my current setup. Can you describe a bit more your experiment and how you observe this error, please?

rshanor, 2022-10-31:
@FrancoisPicardDTI thanks for the reply, and this package!

I am using a GigE camera attached to the end of a robot arm. Nothing is subscribing to the camera topic, I am using the GrabImage action. I am capturing single images, and moving the arm between each grab. My exposure time is pretty long and there is no motion blur, so the issue is not a capture delay but rather the service is returning a stale image. I worked around the problem by adding another call at the start of the action call to grab / flush one extra frame. Let me know if any other info would help.

rshanor, 2022-10-31:
O ya, I am using Ros2.

I can provide the other info later today.

rshanor, 2022-11-08:
Sorry for the delay.

I am using Basler ace acA2440-20gc Color GigE Camera

Here is the bit of code I added right around this line: https://github.com/basler/pylon-ros-camera/blob/galactic/pylon_ros2_camera_component/src/pylon_ros2_camera_node.cpp#L3910
```
  // Work around some off by 1 error I noticed. Grab one frame before grabbing the real
  // frame.
  sensor_msgs::msg::Image img;
  img.encoding = this->pylon_camera_->currentROSEncoding();
  img.height = this->pylon_camera_->imageRows();
  img.width = this->pylon_camera_->imageCols();
  img.step = img.width * this->pylon_camera_->imagePixelDepth();
  this->pylon_camera_->grab(img.data);
```

And here are the params.

```
# Camera parameters

/**:
  ros__parameters:

    #  The tf frame under which the images were published
    camera_frame: camera_frame

    #  The DeviceUserID of the camera. If empty, the first camera found in the
    #  device list will be used
    device_user_id: ""

    #  The CameraInfo URL (Uniform Resource Locator) where the optional intrinsic
    #  camera calibration parameters are stored. This URL string will be parsed
    #  from the ROS-CameraInfoManager:
    camera_info_url: "file:///persist/hardware/calibrated_eoat_camera_cal.yaml"

    #  The encoding of the pixels -- channel meaning, ordering, size
    #  taken from the list of strings in include/sensor_msgs/image_encodings.h
    #  The supported encodings are 'mono8', 'bgr8', 'rgb8', 'bayer_bggr8',
    #  'bayer_gbrg8' and 'bayer_rggb8'
    image_encoding: "bayer_rggb8"

    #  The desired publisher frame rate if listening to the topics.
    #  This parameter can only be set once at startup
    #  Calling the GrabImages-Action can result in a higher framerate
    frame_rate: 5.0

    #  The exposure time in microseconds to be set after opening the camera.
    exposure: 500000.0
```

FrancoisPicardDTI, 2022-11-10:
Hello @rshanor 

Ok, so you have to tell me a bit more. How do you "see" that your one-frame off? How do you visualize it practically speaking? Do you have a dedicated node sending the action's goal, getting the action's results and displaying the acquired images? Or do you do that by command line in a terminal?

I am trying the later until now. Sending the action's goal to trigger the image (with a simple `ros2 action send_goal /my_camera/pylon_ros2_camera_node/grab_images_raw pylon_ros2_camera_interfaces/action/GrabImages "{gain_given: true, gain_values: [2.0]}"`). The action's result being displayed, I can see that an image is acquired and returned. It does not tell me of course if it is one-frame off. How do you do on your side?

FrancoisPicardDTI, 2022-11-15:
Hi @rshanor 
I added in the driver an action client to test this grab image action. Maybe it can help solving your problem? I am not experiencing the offset you are mentioning. Try to see if something is weird on your side with the patch you applied and let me know what you experience.

FrancoisPicardDTI, 2023-01-24:
I am closing this issue for now as it does not seem to be active anymore.

closed 2023-01-24 by FrancoisPicardDTI