basler/pylon-ros-camera #133: Reduce bandwidth when grabbing camera

mbenencase, 2022-11-16:
Hey, I'm using 3 USB cameras on a Xavier AGX device. Two of them are a daA1280-54um and the other is a RealSense T265. All of them use USB 3.0.

Currently, the problem that I'm facing is related to the bandwidth supported by a USB-Hub that we're using. I'd like to know how could I reduce the bandwidth when acquiring images from the daA1280-54um devices.

What we tried so far is:
- Reduce the frame rate of publishing the images (we're using ROS for that).
- Reduce the number of rows and columns of the grabbed image.
- Set the pixel format to MONO8.

There's another possibility to reduce the bandwidth during acquisition?

rows-of-houses, 2022-11-18:
Try to go to Pylon Viewer, select Tools, Bandwidth manager. With this tool you can reduce the bandwidth.

FrancoisPicardDTI, 2023-03-08:
Hello @mbenencase 
Did it solve your issue?

closed 2023-04-24 by FrancoisPicardDTI