basler/pylon-ros-camera #135: New test brakes build on ROS2 rolling

Flova, 2022-11-30:
The newly added test (`test_grab_image_action_client`) breaks the build under ROS2 rolling. I reproduced this issue with the latest commit on multiple machines.

The log of the failing build:
<details>

```
/home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp: In member function ‘void pylon_ros2_camera::TestGrabImageActionClient::send_goal()’:
/home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:77:122: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<pylon_ros2_camera_interfaces::action::GrabImages>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>’} and ‘std::_Bind_helper<false, void (pylon_ros2_camera::TestGrabImageActionClient::*)(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >), pylon_ros2_camera::TestGrabImageActionClient*, const std::_Placeholder<1>&>::type’)
   77 |         send_goal_options.goal_response_callback = std::bind(&TestGrabImageActionClient::goal_response_callback, this, _1);
      |                                                                                                                          ^
In file included from /usr/include/c++/11/future:47,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:18,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:29:
/usr/include/c++/11/bits/std_function.h:530:9: note: candidate: ‘template<class _Functor> std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<_Functor>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = _Functor; _Res = void; _ArgTypes = {std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >}]’
  530 |         operator=(_Functor&& __f)
      |         ^~~~~~~~
/usr/include/c++/11/bits/std_function.h:530:9: note:   template argument deduction/substitution failed:
In file included from /usr/include/c++/11/bits/move.h:57,
                 from /usr/include/c++/11/bits/stl_pair.h:59,
                 from /usr/include/c++/11/bits/stl_algobase.h:64,
                 from /usr/include/c++/11/memory:63,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:153,
                 from /home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:29:
/usr/include/c++/11/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using __enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>&]’:
/usr/include/c++/11/bits/std_function.h:353:8:   required by substitution of ‘template<class _Res, class ... _ArgTypes> template<class _Cond, class _Tp> using _Requires = std::__enable_if_t<_Cond::value, _Tp> [with _Cond = std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>::_Callable<std::_Bind<void (pylon_ros2_camera::TestGrabImageActionClient::*(pylon_ros2_camera::TestGrabImageActionClient*, std::_Placeholder<1>))(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >)>, std::_Bind<void (pylon_ros2_camera::TestGrabImageActionClient::*(pylon_ros2_camera::TestGrabImageActionClient*, std::_Placeholder<1>))(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >)>, std::__invoke_result<std::_Bind<void (pylon_ros2_camera::TestGrabImageActionClient::*(pylon_ros2_camera::TestGrabImageActionClient*, std::_Placeholder<1>))(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >)>&, std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > > >; _Tp = std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>&; _Res = void; _ArgTypes = {std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >}]’
/usr/include/c++/11/bits/std_function.h:530:2:   required by substitution of ‘template<class _Functor> std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>::_Requires<std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>::_Callable<_Functor, typename std::enable_if<(! std::is_same<typename std::remove_cv<typename std::remove_reference<_Tp>::type>::type, std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)> >::value), std::decay<_Tp> >::type::type, std::__invoke_result<typename std::enable_if<(! std::is_same<typename std::remove_cv<typename std::remove_reference<_Tp>::type>::type, std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)> >::value), std::decay<_Tp> >::type::type&, std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > > >, std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>&> std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>::operator=<_Functor>(_Functor&&) [with _Functor = std::_Bind<void (pylon_ros2_camera::TestGrabImageActionClient::*(pylon_ros2_camera::TestGrabImageActionClient*, std::_Placeholder<1>))(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >)>]’
/home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:77:122:   required from here
/usr/include/c++/11/type_traits:2211:11: error: no type named ‘type’ in ‘struct std::enable_if<false, std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >)>&>’
 2211 |     using __enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~~~
In file included from /usr/include/c++/11/future:47,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:18,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:29:
/usr/include/c++/11/bits/std_function.h:540:9: note: candidate: ‘template<class _Functor> std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::reference_wrapper<_Functor>) [with _Functor = _Functor; _Res = void; _ArgTypes = {std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> >}]’
  540 |         operator=(reference_wrapper<_Functor> __f) noexcept
      |         ^~~~~~~~
/usr/include/c++/11/bits/std_function.h:540:9: note:   template argument deduction/substitution failed:
/home/florian/Projekt/bitbots/colcon_ws/src/bitbots_meta/lib/pylon-ros-camera/pylon_ros2_camera_wrapper/test/test_grab_image_action_client.cpp:77:122: note:   ‘std::_Bind<void (pylon_ros2_camera::TestGrabImageActionClient::*(pylon_ros2_camera::TestGrabImageActionClient*, std::_Placeholder<1>))(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >)>’ is not derived from ‘std::reference_wrapper<_Tp>’
   77 |         send_goal_options.goal_response_callback = std::bind(&TestGrabImageActionClient::goal_response_callback, this, _1);
      |                                                                                                                          ^
[... cut ...]
/usr/include/c++/11/bits/std_function.h:501:17: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (pylon_ros2_camera::TestGrabImageActionClient::*)(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<pylon_ros2_camera_interfaces::action::GrabImages> > >), pylon_ros2_camera::TestGrabImageActionClient*, const std::_Placeholder<1>&>::type’ to ‘std::nullptr_t’
  501 |       operator=(nullptr_t) noexcept
      |                 ^~~~~~~~~

```

</details>

The build process worked ultil the last commit.

FrancoisPicardDTI, 2022-12-06:
Hello @Flova 
I just pushed a new humble version. Can you try it please and see if it solves your problem?

timonegk, 2022-12-07:
I can confirm that this issue is fixed with the latest commit.

closed 2023-01-24 by FrancoisPicardDTI